ansys/pyaedt #1059: How can I move more than one face of a 3D object at the same time?

enrique-rc, 2022-04-19:


enrique-rc, 2022-04-19:
Hi, how can I move more than one face of a 3D object with a single instruction, already knowing which face IDs I want to move?

Samuelopez-ansys, 2022-04-20:
Hi @enrique-rc , 

You have 2 options right now.

1) Use the method move_with_object()

You just need to take the faces of an object:
_face_list = o.faces_
Select the face/s you are interested: 
f = o.faces[0]
And perform the operation:
f.move_with_offset(0.1)

This will move one face. In a loop you will move all the faces you needed.

2) For now it is not implemented MoveFaces as it is in the API, where you could add a list with ID Faces. But you could use the AEDT API command directly:
oEditor = aedtapp.modeler.oeditor
#AEDT API Command
oEditor.MoveFaces(
	[
		"NAME:Selections",
		"Selections:="		, "Cylinder8",
		"NewPartsModelFlag:="	, "Model"
	], 
	[
		"NAME:Parameters",
		[
			"NAME:MoveFacesParameters",
			"MoveAlongNormalFlag:="	, True,
			"OffsetDistance:="	, "1mm",
			"MoveVectorX:="		, "0mm",
			"MoveVectorY:="		, "0mm",
			"MoveVectorZ:="		, "0mm",
			"FacesToMove:="		, [421,423]
		]
	])

We will think about including it in the future.

Let me know if it works for your case.

enrique-rc, 2022-04-20:
hi @Samuelopez-ansys ,

In my case, moving faces one by one with method 1 deforms the object in such way that it causes the modeler to fail. I will stick with method 2 then.

Thank you for you support.

Samuelopez-ansys, 2022-04-20:
Hi @enrique-rc ,

I have added a new method, it is in the last main version:

aedtapp.modeler.move_face(face_list, offset)

face_list could be a list or not of Faces ID  or also the type pyaedt.modeler.Object3d.FacePrimitive. Both will work.

I will close the issue. Thanks

closed 2022-04-20 by Samuelopez-ansys