ansys/pyaedt #1064: Wrong Function of edb.core_stackup.stackup_layers.add_layer

linmingchih, 2022-04-20:
I expect to add a new layer, but it replaces the existing layer. Please help to fix it. Thanks.

        from pyaedt import Edb
        
        edb = Edb("D:\demo2\Galileo_G87173_20430.aedb", 'Galileo_G87173_204', edbversion='2022.1')
        edb.core_stackup.stackup_layers.add_layer('xyz', 'UNNAMED_004', 'copper', thickness='0mil')
        
        edb.save_edb_as('d:/demo2/temp.edb')

[image]

svandenb-dev, 2022-05-05:
[image]

@linmingchih  I am not able to reproduce the issue. And I think I am using the same project as you.

ring630, 2022-07-22:
@svandenb-dev , I can reproduce the same problem. The existing layer is overwritten.
[image]

SHF101202021, 2022-08-26:
@ring630 @linmingchih @svandenb-dev  @maxcapodi78 @pyAnsysUser

For this issue, I'm not sure the actual reason. So, let's me explain the modification and deducing reasons for the second

### Modification
Copy the methodolgy of ```EDBLayer.update_layers()```, one uses ```System.Collections.Generic.List[Layer]``` to create the stackup rather than directly using  ```Ansys.Ansoft.Edb.Cell.LayerCollection``` 

### Code
```python
class EDBLayers(object):
    
    
    '''skip'''
    
    @pyaedt_function_handler()
    def add_layer(
        self,
        layerName,
        start_layer=None,
        material="copper",
        fillMaterial="",
        thickness="35um",
        layerType=0,
        negative_layer=False,
        roughness_enabled=False,
        etchMap=None,
    ):
    
        thisLC = self._pedbstackup._active_layout.GetLayerCollection()
        layers = list(list(thisLC.Layers(self._edb.Cell.LayerTypeSet.AllLayerSet)))
        layers.reverse()
        
        el = 0.0
        newLayers = List[self._edb.Cell.Layer]()    
        
        if not layers or not start_layer:
            
            '''Not important in this issue'''
            
        else:
            for lyr in layers:
                if not lyr.IsStackupLayer():
                    # Need to be shifted here for building stackup from bottom to top
                    newLayers.Add(newLayer)
                    continue
                
                if lyr.GetName() == start_layer:
                    newLayer = self._edb.Cell.StackupLayer(
                        layerName,
                        self._int_to_layer_types(layerType),
                        self._get_edb_value(0),
                        self._get_edb_value(0),
                        "",
                    )

                    self._edb_object[layerName] = EDBLayer(newLayer.Clone(), self._pedbstackup)
                    newLayer = self._edb_object[layerName].update_layer_vals(
                        layerName,
                        newLayer,
                        etchMap,
                        material,
                        fillMaterial,
                        thickness,
                        negative_layer,
                        roughness_enabled,
                        self._int_to_layer_types(layerType),
                    )
                    newLayer.SetLowerElevation(self._get_edb_value(el))
                    el += newLayer.GetThickness()
                    
                    newLayers.Add(newLayer)
                    lcNew.AddLayerTop(newLayer)
                    
                newLayer = lyr.Clone()
                newLayer.SetLowerElevation(self._get_edb_value(el))
                el += newLayer.GetThickness()
                ###
                newLayers.Add(newLayer)
                ###
             
        newLayers.reverse()
        lcNew2 = self._edb.Cell.LayerCollection()
        lcNew2.SetMode(thisLC.GetMode())
        lcNew2.AddLayers(newLayers)    

        if not self._active_layout.SetLayerCollection(lcNew2):
            self._logger.error("Failed to set new layers when updating the stackup information.")
            return False
        
        '''skip'''

```

### Deducing reason
Look like that there is something wrong when just using ```Ansys.Ansoft.Edb.Cell.LayerCollection``` (I have no condifence actually)

### Result
#### Example1
[image]
[image]
#### Example2
[image]
[image]

pyAnsysUser, 2022-08-26:
@SHF101202021
In which part of the code does your proposal add the layer to the bottom rather than the top? I am still thinking there should be a sensitivity to whether the layer is added to the top half or bottom half of the stackup, when adding to the top or bottom of the adjacent layer (based on GUI behavior). Reading through this, it is all above my coding ability, however.

SHF101202021, 2022-08-27:
@pyAnsysUser 
replace the ```EDBLayers.add_layers()``` with the following codes and try it
```python
    @pyaedt_function_handler()
    def add_layer(
        self,
        layerName,
        start_layer=None,
        material="copper",
        fillMaterial="",
        thickness="35um",
        layerType=0,
        negative_layer=False,
        roughness_enabled=False,
        etchMap=None,
    ):
        """Add a layer after a specific layer.

        Parameters
        ----------
        layerName : str
            Name of the layer to add.
        start_layer : str, optional
            Name of the layer after which to add the new layer.
            The default is ``None``.
        material : str, optional
            Name of the material. The default is ``"copper"``.
        fillMaterial : str, optional
            Name of the fill material. The default is ``""``.)
        thickness : str, optional
            Thickness value, including units. The default is ``"35um"``.
        layerType :
            Type of the layer. The default is ``0``
            ``0``: Signal layer.
            ``1``: Dielectric layer.
            ``2``: Conducting plane layer.
            ``3``: Airline layer.
            ``4``: Error layer.
            ``5``: Symbol layer.
            ``6``: Measure layer.
            ``8``: Assembly layer.
            ``9``: Silkscreen layer.
            ``10``: Solder Mask layer.
            ``11``: Solder Paste layer.
        negative_layer : bool, optional
            ``True`` when negative, ``False`` otherwise.
        roughness_enabled : bool, optional
            ``True`` if the layer has roughness, ``False`` otherwise.
        etchMap : optional
            Etch value if any. The default is ``None``.

        Returns
        -------
        :class:`pyaedt.edb_core.EDB_Data.EDBLayer`
            Layer Object for stackup layers. Boolean otherwise (True in case of success).
        """
        thisLC = self._pedbstackup._active_layout.GetLayerCollection()
        layers = list(list(thisLC.Layers(self._edb.Cell.LayerTypeSet.AllLayerSet)))
        layers.reverse()
        el = 0.0
        newLayers = List[self._edb.Cell.Layer]()
        
        if not layers or not start_layer:
            
            # Set non-stackup layer first
            for lyr in layers:
                if not lyr.IsStackupLayer():
                    newLayers.Add(lyr.Clone())
                    continue                            
            
[... 76 lines cut ...]
                el += newLayer.GetThickness()
                
                newLayers.Add(newLayer)

        
        newLayers.reverse()
        lcNew2 = self._edb.Cell.LayerCollection()
        lcNew2.SetMode(thisLC.GetMode())
        lcNew2.AddLayers(newLayers)
        
        if not self._active_layout.SetLayerCollection(lcNew2):
        # if not self._active_layout.SetLayerCollection(lcNew):
            self._logger.error("Failed to set new layers when updating the stackup information.")
            return False
        self._update_edb_objects()
        allLayers = [
            i.GetName() for i in list(list(self.layer_collection.Layers(self._edb.Cell.LayerTypeSet.AllLayerSet)))
        ]
        if layerName in self.layers:
            return self.layers[layerName]
        elif layerName in allLayers:
            return True
        return False

```

pyAnsysUser, 2022-08-31:
The default project comes in with top/bottom association on only the top and bottom layers:
[image]
The example project has zero thickness UNNAMED_000 and UNNAMED_012 around these two layers, so the conductor won't be removing from the thickness of the adjacent layer. In the general case, however, for a layer with top/bottom association, this is not true.

When the code walks through the stackup in add_layer() , it uses the lower elevation as a reference for the newly added layer, and increases the lower elevation by the layer's thickness. This will yield an incorrect value if an internal layer is defined as a top/bottom association or if a non-zero thickness soldermask value is used.

Since the example project does not have either of these, I do not know if this contributes to the unpredicted behavior.

ring630, 2022-08-31:
@maxcapodi78 @svandenb-dev I found a good way to enhance this method. I am working on it now.

ring630, 2022-08-31:
@svandenb-dev @maxcapodi78 In case you are already working on a branch, here I share my finding. AddLayerAbove() is a better method to add layer. It adjusts elevation automatically. Below is part of my test code. 

`thisLC = self._pedbstackup._active_layout.GetLayerCollection()
        layers = list(list(thisLC.Layers(self._edb.Cell.LayerTypeSet.AllLayerSet)))

        lcNew = self._edb.Cell.LayerCollection()

        flag_first_layer = True
        for lyr in layers:
            if not lyr.IsStackupLayer():
                continue
            lyr_clone = lyr.Clone()
            lyr_name = lyr.GetName()
            if flag_first_layer:
                lcNew.AddLayerTop(lyr_clone)
                flag_first_layer = False
            else:
                lcNew.AddLayerAbove(lyr_clone, lyr_name)

        lcNew.AddLayerAbove(newLayer, start_layer)`

Please let me know if you want me to create a PR.

ring630, 2022-08-31:
@linmingchih @SHF101202021 @pyAnsysUser , there is a new method insert_layer_above(). Please use it to insert layer into existing stackup.

pyAnsysUser, 2022-09-09:
@ring630 I haven't done a ton of testing with this new method, but it appears to work on the example project, however I still get odd behavior on my project (primitives are moved to dielectric layers). I think the next thing I will do is look at an example edb project that has non-zero soldermask. In general, however, I think this functionality is much improved. Is there an improvement based on this issue that can be made around the unit tests? The add_layer() test was minimal and just verified the function could be called, not that a layer was added correctly.

ring630, 2022-09-09:
@pyAnsysUser please describe the problem you have in details with pictures.

pyAnsysUser, 2022-09-11:
@ring630 I was mistaken, there is a warning about primitives on dielectrics, but the sim was erroring due to a bug on my side. This passes the few test cases I've thrown at it.

closed 2022-09-21 by ring630