ansys/pyaedt #1077: [Maxwell2d] Example for parametrics add_calculation

BLeistr, 2022-04-22:
Could you add an example of a parameter study for a magnetostatic analysis under 
https://aedtdocs.pyansys.com/examples/01-Modeling-Setup/Optimetrics.html?

For add_calculation, the parameter 'ranges' must be set, but it is not needed in magnetostatic calculations, in my opinion. 
I have already tried the following:
sweep = m3d.parametrics.add("mag_x", 1, 5, 2, variation_type='LinearStep', parametricname = "Parametric")
sweep.add_calculation(calculation="Bz_mean", ranges = [], report_type="Fields")

This does not cause an error in Python, but gives an error in maxwell:
[image]

Samuelopez-ansys, 2022-04-25:
Hi @BLeistr ,

I don't know exactly the context of this setup, but I guess the issue is if you should specify in ranges the Time you want to calculate, for example:

sweep.add_calculation(calculation=""Bz_mean", ranges={"Time": "1s"})

Please add a few lines of code to reproduce a dummy geometry and setup, and I will be able to help you more in detail.

BLeistr, 2022-04-26:
Thank you for your feedback. Here is a short simplified example:
---------------------------------
from pyaedt import Maxwell3d
m3d = Maxwell3d()

# Create Variables
m3d["mag_x"] = "5mm"
m3d["mag_y"] = "10mm"
m3d["mag_z"] = "20mm"
   
# Create Magnet
magnet = m3d.modeler.primitives.create_box(["-mag_x/2", "-mag_y/2", "-mag_z/2"], ["mag_x", "mag_y", "mag_z"], name="Magnet", matname="NdFe30")
region = m3d.modeler.create_air_region(200,200,200,200,200,200)

# Create Setup 
Setup = m3d.create_setup("Setup1")

# Field calculator
m3d.post.ofieldsreporter.EnterQty("B")
m3d.post.ofieldsreporter.CalcOp("ScalarX")
m3d.post.ofieldsreporter.AddNamedExpression("Bx", "Fields")

# Example for the calculation of a volume average
m3d.post.ofieldsreporter.CopyNamedExprToStack("Bx")
m3d.post.ofieldsreporter.EnterVol("Magnet")
m3d.post.ofieldsreporter.CalcOp("Integrate")
m3d.post.ofieldsreporter.EnterScalar(1)
m3d.post.ofieldsreporter.EnterVol("Magnet")
m3d.post.ofieldsreporter.CalcOp("Integrate")
m3d.post.ofieldsreporter.CalcOp("/")
m3d.post.ofieldsreporter.AddNamedExpression("Bx_mean", "Fields")

# Parametrics
sweep = m3d.parametrics.add("mag_x", 1, 5, 2, variation_type='LinearStep', parametricname = "Parametric")
sweep.add_calculation(calculation="Bx_mean", ranges = [], report_type="Fields")
-------------------------------------
For magnetostatic calculations there is, in my opinion, no specification for a range.

BLeistr, 2022-04-26:
Here the source code, if one adds it over the surface. The entry ranges remains free.  

oModule.EditSetup("Parametric", 
	[
		"NAME:Parametric",
		"IsEnabled:="		, True,
		[
			"NAME:ProdOptiSetupDataV2",
			"SaveFields:="		, False,
			"CopyMesh:="		, False,
			"SolveWithCopiedMeshOnly:=", True
		],
		[
			"NAME:StartingPoint"
		],
		"Sim. Setups:="		, ["Setup1"],
		[
			"NAME:Sweeps",
			[
				"NAME:SweepDefinition",
				"Variable:="		, "mag_x",
				"Data:="		, "LIN 1mm 5mm 2mm",
				"OffsetF1:="		, False,
				"Synchronize:="		, 0
			]
		],
		[
			"NAME:Sweep Operations"
		],
		[
			"NAME:Goals",
			[
				"NAME:Goal",
				"ReportType:="		, "Fields",
				"Solution:="		, "Setup1 : LastAdaptive",
				[
					"NAME:SimValueContext"
				],
				"Calculation:="		, "Bx_mean",
				"Name:="		, "Bx_mean",
				[
					"NAME:Ranges"
				]
			]
		]
	])

Samuelopez-ansys, 2022-04-27:
Hi,

Ranges are needed because if you check the recorded script from the API, you will find Ranges are there (but empty):
[image]

But I found a couple of bugs I fixed for Maxwell and Optimetrics. If you update to the latest main, you will find your script works, please let me know if it works to close the issue.

BLeistr, 2022-04-27:
Thank you very much. Yes, now it works.

closed 2022-04-28 by Samuelopez-ansys