ansys/pyaedt #1078: [Maxwell2D] Determination of the return path for matrix definition

BLeistr, 2022-04-22:
How to define the return path for a 2D (XY) coil turn when using assign_matrix?
The function itself has no corresponding property.

[image]

BLeistr, 2022-04-28:
An example ist:

from pyaedt import Maxwell2d

#% Definition von Variablen in mm!
mx = 10     # magnet length in x-direction
my = 3      # magnet length in y-direction
p = 24      # phase length - distance between center of 3 magents
dist = 5  # distance between two magnets in z-direction
cx = 8      # coil length in x-direction
cy = 3   # coil length in y-direction
pc = 18     # phase length - distance between center of 3 coil
dx = 0      # displacement of coils
m_num = 2   # number of magnets per line
c_num = 2   # number of coil strand

m2d = Maxwell2d(solution_type="MagnetostaticXY")
   
#% Create Variables
exec('m2d["mx"] = "'+str(mx)+'mm"')
exec('m2d["my"] = "'+str(my)+'mm"') 
exec('m2d["p"] = "'+str(p)+'mm"') 
exec('m2d["dist"] = "'+str(dist)+'mm"') 
exec('m2d["cx"] = "'+str(cx)+'mm"') 
exec('m2d["cy"] = "'+str(cy)+'mm"') 
exec('m2d["pc"] = "'+str(pc)+'mm"') 
exec('m2d["dx"] = "'+str(dx)+'mm"') 
exec('m2d["m_num"] = "'+str(m_num)+'"') 
exec('m2d["c_num"] = "'+str(c_num)+'"')

# Create magnetic materials in Maxwell
Hc = -840000 # coercitivity in A/m
m2d.materials.checkifmaterialexists("NdFe30")
mat1 = m2d.materials.duplicate_material("NdFe30", "ndfe_py")
mat1.set_magnetic_coercitivity(Hc,0,1,0)
mat1.update()
mat2 = m2d.materials.duplicate_material("NdFe30", "ndfe_ny")
mat2.set_magnetic_coercitivity(Hc,0,-1,0) 
mat2.update()

#% Create Magnet --> [x size, y size]
for n in range(0, m_num):
    mat = "ndfe_ny" if n % 2 == 0 else "ndfe_py" # change of magnetization direction
    for nn in range(-1,2,2):
        mag_name = "Magnet_"+str(n+1) if nn ==1 else "Magnet_m"+str(n+1)
        m2d.modeler.primitives.create_rectangle(["-(m_num-1)*p/4-mx/2+"+str(n)+"*p/2", str(nn)+"*dist/2", 0], ["mx", str(nn)+"*my"],is_covered=True, name=mag_name, matname=mat)
for n in range(0, c_num):
    coil_name = "Coil_"+str(n+1)
    coil = m2d.modeler.primitives.create_rectangle(["-(c_num-1)*pc/4-cx/2+"+str(n)+"*pc/2+dx", "-cy/2", 0], ["cx", cy"],is_covered=True, name=coil_name, matname="vacuum")

#% Create Air Region
region = m2d.modeler.primitives.create_region([100,300,100,300])
#% Create Boundaries
m2d.assign_balloon(region.edges, bound_name = "Balloon") 

#% Assign current on the coils (direction: true -> positive, false -> negative)
m2d.assign_current("Coil_1", amplitude=1, swap_direction=False, name="Current1")
m2d.assign_current("Coil_2", amplitude=1, swap_direction=True, name="Current2")

I want to calculate the inductance:
[image]

I tried something like:
L = m2d.assign_matrix('Current1', matrix_name='Matrix')
L.props['ReturnPath' = 'Current2

But L has no attribute 'props'.

Samuelopez-ansys, 2022-04-28:
Hi, 

It is not still implemented in pyAEDT, but you could use AEDT API to perform this operation:

m2d.o_maxwell_parameters.EditMatrix(L, ["NAME:Matrix",["NAME:MatrixEntry",["NAME:MatrixEntry", "Source:=", "Current1", "NumberOfTurns:=", "2","ReturnPath:=", "Current2"]],["NAME:MatrixGroup"]])

Let me know if it works.

Regards,
Samuel

BLeistr, 2022-05-02:
Hello, 
thank you very much, this solution also works.

BLeistr, 2022-05-03:
But is there already a way to pass matrix solutions to Python directly, via a get command?
Or is it mandatory to go via create_report and export_report_to_file and then read the file?

closed 2022-05-06 by Samuelopez-ansys