ansys/pyaedt #1080: How to change some parameters in maxwell2d

tako861225, 2022-04-24:
How to change the parameters in Excitations if there is already assign value before ?
1. number of conductors 
[image]
I tried to reassign , but it not work .
[image]
2. parameters in winding 
[image]
I would like to change the type and others , but I don't know how QQ
I tried to reassign , but it not work too.

thanks lot 
Evelyn

tako861225, 2022-04-25:
I try the update instruction. It will return "True" but nothing change , both ``assign_winding`` and ``assign_coil``
[image]

Samuelopez-ansys, 2022-04-25:
Hi Evelyn,

Which version of pyaedt are you using?

I have tested the winding in the last pyaedt release, and it works:

bounds = aedtapp.assign_winding(current_value=20e-3, coil_terminals=["Coil"])
bounds.props["Type"] = "Voltage"
bounds.props["Current"] = "0.01A"

[image]

You don't need bounds.update() anymore, it is automatically updated.

tako861225, 2022-04-25:
Thanks for the respond ❤️ 
I use the latest version of pyaedt.
I notice that props.[""] should use name on GUI ,not the functions parameters name.
 (I change it from "winding_type" to "Type")
But in assing_coil ,the function will return ``CoilObject``  that I couldn't find the explanation about it.
 
[image]

I try the same way to update the parameters but it doesn't work .
[image]
[image]

Samuelopez-ansys, 2022-04-25:
Hi Evelyn,

This is not the way of using this method.

1. You should put a list as input_object, as you attached, input_object is a list of objects or face IDs:
        bounds = aedtapp.assign_coil(input_object=["Coil"])

2. The properties could have different names because internally the method could have different output properties, so you should check the different output properties:
[image]

Then you can play with them:
bounds.props["Conductor number"] = "3"
bounds.props["PolarityType"] = "Negative"

Let me know if it is clear and if it works in your case.

closed 2022-04-25 by Samuelopez-ansys