ansys/pyaedt #1184: Issue with get_all_sources API

Vamsi-Yaddanapudi, 2022-05-16:
Hello,

While using get all sources in PyAEDT for Q3D project, it fails.

Can you please check.

Below is log message,

/////////////
Error in :   File "C:\Users\ykrishna\AppData\Local\Programs\Python\Python37\lib\site-packages\pyaedt\application\Analysis3D.py", line 513, in get_all_sources
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=get_all_sources
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=get_all_sources
///////////////////////////

Thanks,
Vamsi

Samuelopez-ansys, 2022-06-16:
Hi @Vamsi-Yaddanapudi ,

There is a Typo in Q3D Documentation, the method GetAllSources is not available anymore from:

oModule = oDesign.GetModule("Solutions")
oModule.GetAllSources() -----> Error.

Please could you try and submit a bug in Ansys development platform? If it is located in other place of the API let us know and we will modify it.

The workaround is using directly:
app = Q3d()
app.excitations -----> This will return the Q3D Excitations:

1 = {str} 'Box1_1:Source2'
0 = {str} 'Box1:Source1'

Please let me know if it is clear and close the Issue.

maxcapodi78, 2022-06-30:
@Samuelopez-ansys maybe could be useful to override the get_all_sources method of Q3d with excitations. in this way it won't fail and will return anyway useful info. What do you think?

closed 2022-06-30 by Samuelopez-ansys