ansys/pyaedt #1238: Please update Maxwell example: Maxwell 2d: Magnet Transient Analysis

tizianrot, 2022-06-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Please update the example with the following requests:
1) add this line when design settings are set: 
    M2D.change_symmetry_multiplier("SymmetryFactor")

2) please change the name to:
     PM synchronous motor transient analysis

Many thanks for your help,
Warm Regards,
Tiziana

### Steps To Reproduce

it is just an update

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

not relevant

gmalinve, 2022-06-20:
In Maxwell2D_NissanLeaf.py:
- Change example title at line 2.
- Add at line 457: M2D.change_symmetry_multiplier("SymmetryFactor")

iaskgithub, 2023-09-16:
filename_lam, filename_PM = pyaedt.downloads.download_leaf()  in line 164 ,seems that will leads to an error ,how to fix it ?

iaskgithub, 2023-09-16:
> filename_lam, filename_PM = pyaedt.downloads.download_leaf() in line 164 ,seems that will leads to an error ,how to fix it ?

in Nissan leaf example

gmalinve, 2023-09-16:
Hi @iaskgithub, 

Which error do you get? and which version of pyaedt are you using?

Giulia

iaskgithub, 2023-09-17:
> Hi @iaskgithub,
> 
> Which error do you get? and which version of pyaedt are you using?
> 
> Giulia
thank you,i have solve that problem,it's not a bug of the pyaedt