ansys/pyaedt #1241: Unable to get the sweep units of get_solution_data_per_variation method

mahesh-dhanekula, 2022-06-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I have been using pyaedt v0.4.57 till yesterday and updated to latest version today.

I am using ```get_solution_data_per_variation()``` method to get spectral data of transient simulation from Circuit.

Due to some refactoring in latest version of PyAEDT, my code is throwing errors as I am referring to properties that were no longer part of this refactored method.  

```data = circuit.post.get_solution_data_per_variation(soltype='Standard', setup_sweep_name=setup_name, ctxt=context, sweeps={'Spectrum':'All'}, expression=exp)```

To get the sweep data and units following is the code I have used

```data = data.sweeps['Spectrum']```

This is throwing error as there is no attribute named sweeps.

```units = data.units_sweeps['Spectrum']```

This is throwing error as there is no key in sweeps named "Spectrum"

Can someone confirm if this is issue in PyAEDT or am I doing something wrong ?

### Steps To Reproduce

```py
context = ["NAME:Context", "SimValueContext:=", [2,0,2,0,False,False,1,1,0,1,1,"",0,0, "CG",False,"1","CP",False,"0","IT",False,"1","KP",False,"0","MF",False,"100MHz", "NUMLEVELS",False,"0","TE",False,"100us","TS",False,"0ns","WT",False,"4","WW",False,"100"]]

data = circuit.post.get_solution_data_per_variation(soltype='Standard', setup_sweep_name="MyTransient", ctxt=context, sweeps={'Spectrum':'All'}, expression="V(3)")

units = data.units_sweeps['Spectrum']
```
This is giving me invalid key error as there is no key named "Spectrum"

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

aiohttp==3.7.4.post0
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
async-timeout==3.0.1
attrs==21.2.0
autobahn==21.3.1
Automat==20.2.0
Babel==2.10.1
backcall==0.2.0
beautifulsoup4==4.10.0
bleach==5.0.0
certifi==2021.10.8
cffi==1.15.0
chardet==4.0.0
charset-normalizer==2.0.7
click==7.1.2
colorama==0.4.4
constantly==15.1.0
cPython==0.0.6
cryptography==35.0.0
cycler==0.10.0
Cython==0.29.28
debugpy==1.6.0
decorator==5.1.0
defusedxml==0.7.1
entrypoints==0.4
fastjsonschema==2.15.3
fpdf2==2.5.4
hyperlink==21.0.0
idna==3.3
imageio==2.9.0
importlib-metadata==4.11.4
importlib-resources==5.7.1
incremental==21.3.0
ipykernel==6.13.0
ipython==7.28.0
ipython-genutils==0.2.0
jedi==0.18.0
Jinja2==3.1.2
json5==0.9.8
jsonschema==4.5.1
jupyter-client==7.3.1
jupyter-core==4.10.0
jupyter-server==1.17.0
jupyterlab==3.4.2
jupyterlab-pygments==0.2.2
jupyterlab-server==2.14.0
kiwisolver==1.3.2
MarkupSafe==2.1.1
matplotlib==3.4.3
matplotlib-inline==0.1.3
meshio==4.4.6
mistune==0.8.4
multidict==5.2.0
nbclassic==0.3.7
nbclient==0.6.3
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
networkx==2.7.1
notebook==6.4.11
notebook-shim==0.1.0
numpy==1.21.3
packaging==21.3
pandocfilters==1.5.0
parso==0.8.2
pickleshare==0.7.5
Pillow==9.1.1
plumbum==1.7.0
prometheus-client==0.14.1
prompt-toolkit==3.0.21
psutil==5.9.1
pyaedt==0.4.73
pycodestyle==2.8.0
pycparser==2.20
Pygments==2.10.0
pymongo==3.12.1
pyparsing==2.4.7
pypiwin32==223
PyQt5==5.15.4
pyqt5-plugins==5.15.4.2.2
PyQt5-Qt5==5.15.2
PyQt5-sip==12.9.0
pyqt5-tools==5.15.4.3.2
PyQt5Designer==5.14.1
pyrsistent==0.18.1
python-dateutil==2.8.2
python-dotenv==0.19.2
pythonnet==2.5.2
pytz==2022.1
pyvista==0.34.1
PyWavelets==1.2.0
pywin32==303
pywinpty==2.0.5
pyzmq==23.0.0
qt5-applications==5.15.2.2.2
qt5-tools==5.15.2.1.2
requests==2.26.0
rpyc==5.0.1
scikit-image==0.19.2
scipy==1.7.3
scooby==0.5.7
Send2Trash==1.8.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.1
terminado==0.15.0
tifffile==2022.2.9
tinycss2==1.1.1
tornado==6.1
traitlets==5.1.0
Twisted==21.7.0
twisted-iocpsupport==1.0.2
txaio==21.2.1
typing-extensions==3.10.0.2
urllib3==1.26.7
vtk==9.0.3
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.2
wget==3.2
wslink==1.1.0
yarl==1.7.0
zipp==3.8.0
zope.interface==5.4.0

Samuelopez-ansys, 2022-06-16:
Hi @mahesh-dhanekula , the method has been improved in the last releases.

Now you should just use:
data1 = circuit_test.post.get_solution_data(expressions="dB(V(net_11))", setup_sweep_name="Transient", domain="Spectrum")
In data, you will have all the values, for example, to access the Y values: data1.data_magnitude() or the sweep using data1.primary_sweep

You could create the report using:
new_report = circuit_test.post.reports_by_category.spectral(["dB(V(net_11))"], "Transient")
new_report.window = "Hanning"
new_report.max_freq = "1GHz"
new_report.time_start = "1ns"
new_report.time_stop = "190ns"
new_report.plot_continous_spectrum = True

Let me know if it works for you and close the issue.

Regards

Samuelopez-ansys, 2022-06-20:
Hi @mahesh-dhanekula ,

Now you should be able to get the units sweep and perform the correct report setup:

context = {"adjust_coherent_gain": False, "plot_continous_spectrum": False, "algorithm": "FFT", "kaisser_coeff": 0,
           "max_frequency": "100MHz", "time_stop": "2.5us", "time_start": "0ps", "window": "Rectangular"}

spectralPlotData = app.post.get_solution_data(expressions="V(Vprobe1)", setup_sweep_name="EMItransient",
                                              primary_sweep_variable="Spectrum", domain="Spectral", context=context)

spectralPlotData.units_sweeps

maxcapodi78, 2022-06-27:
@mahesh-dhanekula  we close the issue as it is solved. Please reopen if needed

closed 2022-06-27 by maxcapodi78