ansys/pyaedt #1266: plot(some expression) applies mag() to the expression.

zebanaqviWAT, 2022-06-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

x = hfss.post.get_solution_data()
x.plot()
Plot() always applies mag() to the expression.  

I think the behavior should be that if the expression is already real quantity, plot it as it is. If it is complex, then we could do either 1- default to mag or 2- plot both real and imag in separate or same plot 3- or just error saying it is complex.

[image]
[image]

### Steps To Reproduce

use the plot function in pyaedt as shown in description. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

PS D:\GUIenv\PhaseCenter_GUI> python -m pip freeze
aiohttp==3.8.1
aiosignal==1.2.0
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asttokens==2.0.5
async-timeout==4.0.2
attrs==21.4.0
Babel==2.10.1
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.0
certifi==2022.5.18.1
cffi==1.15.0
charset-normalizer==2.0.12
colorama==0.4.4
cycler==0.11.0
debugpy==1.6.0
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
executing==0.8.3
fastjsonschema==2.15.3
fonttools==4.33.3
frozenlist==1.3.0
idna==3.3
imageio==2.19.3
importlib-metadata==4.11.4
ipykernel==6.13.0
ipython==8.4.0
ipython-genutils==0.2.0
jedi==0.18.1
Jinja2==3.1.2
json5==0.9.8
jsonschema==4.6.0
jupyter-client==7.3.1
jupyter-core==4.10.0
jupyter-server==1.17.0
jupyterlab==3.4.2
jupyterlab-pygments==0.2.2
jupyterlab-server==2.14.0
kiwisolver==1.4.2
MarkupSafe==2.1.1
matplotlib==3.5.2
matplotlib-inline==0.1.3
mistune==0.8.4
multidict==6.0.2
nbclassic==0.3.7
nbclient==0.6.4
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
notebook==6.4.11
notebook-shim==0.1.0
numpy==1.22.4
packaging==21.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.1.1
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.29
psutil==5.9.1
pure-eval==0.2.2
pyaedt==0.4.80
pycparser==2.21
Pygments==2.12.0
pyparsing==3.0.9
pyrsistent==0.18.1
PySide6==6.3.0
PySide6-Addons==6.3.0
PySide6-Essentials==6.3.0
python-dateutil==2.8.2
pythonnet @ https://github.com/pyansys/PyAEDT/raw/release/0.3/pythonnet-2.5.2-cp39-cp39-win_amd64.whl
pytz==2022.1
pyvista==0.34.1
pywin32==304
pywinpty==2.0.5
pyzmq==23.1.0
requests==2.27.1
rpyc==5.0.1
scooby==0.5.12
Send2Trash==1.8.0
shiboken6==6.3.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.2.post1
stack-data==0.2.0
terminado==0.15.0
tinycss2==1.1.1
tornado==6.1
traitlets==5.2.2.post1
urllib3==1.26.9
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.2
wslink==1.6.5
yarl==1.7.2
zipp==3.8.0
PS D:\GUIenv\PhaseCenter_GUI> python -m pip freeze

MaxJPRey, 2022-06-14:
Hi @zebanaqviWAT ,
Have you tried with the `math_formula` argument.
It is made for what you are asking I think.

```
x = hfss.post.get_solution_data()
x.plot(math_formula="mag")
x.plot(math_formula="re")
x.plot(math_formula="im")
```

You can precisely decide what to plot with the curve.

Thanks for the feedback.

zebanaqviWAT, 2022-06-14:
[image]
Thanks for your reply @MaxJPRey. Yes I did. In PostProcessor.py it is forcing mag() to be applied. I do not want to apply any math formula since the expression already contains it - cang_deg(rEPhi). There may be more users who may not want to apply any formula (like choosing None). If they don't specify it, it will apply mag(). One may try no expression but I was just hoping it would accept expression as it is to preserve the negative values. 
[image]

maxcapodi78, 2022-06-30:
@zebanaqviWAT  @MaxJPRey correct point. I think we should consider also a default with no formula.

maxcapodi78, 2022-06-30:
@zebanaqviWAT I've fixed it. now if you don't apply any formula then it will plot the real data in solution data.