ansys/pyaedt #1325: After creating a polyline, I would like to fillet a selection of the vertices.

priti-ashvin-shah-ibm, 2022-06-22:
### Description of the feature

When performing fillet, I would like to select the vertices of polyline which I can fillet.  If there is another way to fillet a selection of vertices for a polyline, provide documentation to do so. 

### Steps for implementing the feature

```

import pyaedt
from pyaedt import Desktop
from pyaedt import Q3d

# https://aedtdocs.pyansys.com/API/_autosummary/pyaedt.Desktop.html
# Don't need to add the version: Version of AEDT to use.
# The default is None, in which case the active setup or latest installed version is used.
desktop = Desktop(specified_version=None,
                  non_graphical=False,
                  new_desktop_session=True,
                  close_on_exit=False,
                  student_version=False)

q3d = Q3d()
q3d_modeler = q3d.modeler

# Confirm the base units of Ansys is mm.
q3d.modeler.model_units = "mm"
q3d.modeler.oeditor.SetModelUnits(
    ["NAME:Units Parameter", "Units:=", "mm", "Rescale:=", False])

points_3d = [[0.425, 0.195, 2.0], [0.725, 0.195, 2.0],
             [0.725, -0.038285413, 2.0], [0.925, -0.038285413, 2.0],
             [0.925, 0.428285413, 2.0], [1.125, 0.428285413, 2.0],
             [1.125, -0.038285413, 2.0], [1.325, -0.038285413, 2.0],
             [1.325, 0.428285413, 2.0], [1.525, 0.428285413, 2.0],
             [1.525, 0.0, 2.0], [1.75, 0.0, 2.0]]

a_polyline = q3d.modeler.create_polyline(points_3d,
                                         name='trace_readout',
                                         cover_surface=False,
                                         close_surface=False,
                                         matname='pec')

q3d.modeler['trace_readout'].transparency = 0.0

qc_fillet = 8.999999999999999e-05

PRECISION = 9

idxs_to_fillet = [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

if idxs_to_fillet:
    vertices = a_polyline.vertices

    # All primitives are in objects.
    #self.q3d.modeler.objects[a_polyline.id] == a_polyline

    #Don't know why these are equal. What is the difference
    #a_polyline.points == a_polyline.vertex_positions

    vp = pyaedt.modeler.Object3d.VertexPrimitive(a_polyline, a_polyline.id)

    if isinstance(idxs_to_fillet, list):
        to_fillet = [vertices[v] for v in idxs_to_fillet]

    ##### I would think that I could used ###########
    #a_polyline.fillet(selection=to_fillet, radius=qc_fillet, setback=0.0)

    ######## Tried the two below with ERROR message shown below.  ##########
    #vp.fillet(radius=qc_fillet, setback=0.0)

    # ERROR message:
    # pyaedt error: Operation failed, generating an unclassified object. Check and retry.
    # Operation failed, generating an unclassified object. Check and retry.
    # False

    #self.q3d.modeler._fillet(qc_fillet, idxs_to_fillet, a_polyline)
desktop.release_desktop(close_projects=False, close_on_exit=False)
```

priti-ashvin-shah-ibm, 2022-06-22:
Within pyepr repo, the comm port was used, not pyaedt.  If it would be helpful, the command is shown below.

```
self._modeler.Fillet(["NAME:Selections", "Selections:=", obj], [
             "NAME:Parameters",
             [
                 "NAME:FilletParameters", "Edges:=", [], "Vertices:=",
                 to_fillet, "Radius:=", radius, "Setback:=", "0mm"
             ]
         ])
```

maxcapodi78, 2022-06-22:
@priti-ashvin-shah-ibm 
We implemente the same method to the property of edges and lines. I will make sure there is also at modeler level so you can use and assign multiple objects.
Anyway the following code does what you need. Consider that the fillet add a point to the line, that's why the original indexing is not valid. 
you can put it after the code you made to create the polyline

`
idxs_to_fillet = [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]
k=0
for i in idxs_to_fillet:
     a_polyline.vertices[i+k].fillet(radius=qc_fillet)
     k+=1
`

Samuelopez-ansys, 2022-06-27:
Hi @priti-ashvin-shah-ibm ,

Did it solve the issue?

Regards