ansys/pyaedt #1340: Q3D: reduced matrix operations on already reduced matrices

tizianrot, 2022-06-29:
### Description of the feature

Pften it is usaful to cascade different Matrix operation reductions, and the option is the Q3D GUI, like "create another join in Series". Up to now matrix reductions work (as far as I know) only on the Original matrix

[image]

### Steps for implementing the feature

It could be another input argument for the function:
insert_reduced_matrix(self, operation_name, source_names=None, rm_name=None)

such as the matrix (original or already reduced) on which we want apply this further matrix operation

### Useful links and references

Please see the training material on Q3D Module M=5 matrix reductions

gmalinve, 2022-07-06:
Hi @tizianrot , I just checked and if I'm not mistake we have a method called add_operation which accepts as inputs the operation type, taken from this enumeration:
`class MATRIXOPERATIONSQ3D(object):
    """Matrix Reduction types."""

    (JoinSeries, JoinParallel, FloatNet, GroundNet, FloatTerminal, FloatInfinity, ReturnPath, AddSink, MoveSink) = (
        "JoinSeries",
        "JoinParallel",
        "FloatNet",
        "GroundNet",
        "FloatTerminal",
        "FloatInfinity",
        "ReturnPath",
        "AddSink",
        "MoveSink",
    )`
and a list of source names.
This will add a new operation to an existing matrix.
Hopefully this is what you were referring to!

Samuelopez-ansys, 2022-09-01:
Hi @tizianrot ,

Is the issue solved as @gmalinve commented?

Please close the issue if it is solved

gmalinve, 2022-10-03:
Hi @tizianrot ,

Any feedback on this issue?
did you try what I suggested in previous comment?

closed 2022-10-27 by Samuelopez-ansys