ansys/pyaedt #1350: Bug located in PostProcessing.PostProcessor.create_fieldplot_surface()

obrienpja, 2022-07-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to plot the electric vector field for a device using pyaedt, like how I have plotted it with the GUI as shown in the following picture:
[image]
I cannot find the proper API call to plot the electric vector field, so I am wondering if there is a bug with AdvancedPostProcessing.PostProcessor.create_fieldplot_surface(). I get the following error when trying to create the fieldplot:
design.post.create_fieldplot_surface(objlist=[design._modeler.object_list[0].faces[0]], quantityName='Vector_E1', setup_name='Setup')
pyaedt error: Method create Failed:  General or AEDT Error. Please Check again
Method create Failed:  General or AEDT Error. Please Check again
pyaedt error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024381), None)
(-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024381), None)
pyaedt error: Project: pyaedt-sample, Design: Readout (Eigenmode), [error] Script macro error: Plot definition data is not valid for this defintion type. (11:08:02 AM  Jul 01, 2022)

Project: pyaedt-sample, Design: Readout (Eigenmode), [error] Script macro error: Plot definition data is not valid for this defintion type. (11:08:02 AM  Jul 01, 2022)

pyaedt error: Arguments Provided: 
Arguments Provided: 
pyaedt error: Error in :   File "C:\Users\patrick.obrien\AppData\Roaming\Python\Python39\site-packages\pyaedt\modules\PostProcessor.py", line 1811, in create
Error in :   File "C:\Users\patrick.obrien\AppData\Roaming\Python\Python39\site-packages\pyaedt\modules\PostProcessor.py", line 1811, in create
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create
False
pydevd warning: Getting attribute FacePrimitive.normal was slow (took 0.23s)

The key part here is that no matter what I try, I am always getting the error:
**Project: pyaedt-sample, Design: Readout (Eigenmode), [error] Script macro error: Plot definition data is not valid for this defintion type. (11:08:02 AM  Jul 01, 2022)**

### Steps To Reproduce

Create an HFSS Eigenmode design, and solve for a mode, then run the following script:

import pyaedt
import os

desktopVersion = "2021.2"
NewThread = False
non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "False").lower() in ("true", "1", "t")
desktop = pyaedt.Desktop(desktopVersion, non_graphical=non_graphical, new_desktop_session=NewThread)

design = pyaedt.Hfss('pyaedt-sample', 'Readout')
design.post.create_fieldplot_surface(objlist=[design._modeler.object_list[0].faces[0]], quantityName='Vector_E1', setup_name='Setup')

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

addict @ file:///home/conda/feedstock_root/build_artifacts/addict_1636818143388/work
aiohttp==3.8.1
aiosignal==1.2.0
alabaster==0.7.12
anyio==3.6.1
appdirs==1.4.4
argon2-cffi @ file:///home/conda/feedstock_root/build_artifacts/argon2-cffi_1640817743617/work
argon2-cffi-bindings @ file:///D:/bld/argon2-cffi-bindings_1640855319298/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1618968359944/work
async-timeout==4.0.2
attrdict==2.0.1
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1640799537051/work
Babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1619719576210/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
[... 133 lines cut ...]
sphinxcontrib-serializinghtml @ file:///home/conda/feedstock_root/build_artifacts/sphinxcontrib-serializinghtml_1637000050264/work
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1644872665635/work
statsmodels @ file:///D:/bld/statsmodels_1644535868573/work
sympy @ file:///D:/bld/sympy_1646599698375/work
terminado @ file:///D:/bld/terminado_1646684769523/work
testpath @ file:///home/conda/feedstock_root/build_artifacts/testpath_1645693042223/work
threadpoolctl @ file:///home/conda/feedstock_root/build_artifacts/threadpoolctl_1643647933166/work
tinycss2==1.1.1
tornado @ file:///D:/bld/tornado_1635819745510/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1646031859244/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1635260543454/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1644850595256/work
unicodedata2 @ file:///D:/bld/unicodedata2_1640031227379/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1641584929973/work
vtk==9.1.0
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1600965781394/work
webencodings==0.5.1
websocket-client==1.3.2
widgetsnbextension @ file:///D:/bld/widgetsnbextension_1637174379490/work
win-inet-pton @ file:///D:/bld/win_inet_pton_1635832858406/work
wslink==1.6.5
xarray @ file:///home/conda/feedstock_root/build_artifacts/xarray_1646254587173/work
xyzservices @ file:///home/conda/feedstock_root/build_artifacts/xyzservices_1646849493349/work
yarl==1.7.2
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1643828507773/work

obrienpja, 2022-07-01:
While this is an issue involving create_fieldplot_surface(), I was wondering if it is possible to get error messages that are more specific? I am happy to open another issue on that if necessary. We are really looking forward to migrating our code to pyaedt, and giving better error messages will facilitate this migration. Thank you!

maxcapodi78, 2022-07-19:
Hi @obrienpja this is not a bug.  The working method should be

"""
design.post.create_fieldplot_surface(objlist=[design._modeler.object_list[0].faces[0]], quantityName='Vector_E', setup_name='Setup')
"""

The quantity to plot is Vector_E. Vector_E1 is the name of the plot given by default by AEDT.
Regarding your second topic, yes, please feel free to open a discussion, we always are open to improve the code.

obrienpja, 2022-07-19:
I will try the syntax you posted. Thank you very much!

closed 2022-07-19 by obrienpja

maxcapodi78, 2022-07-28:
@obrienpja have you tried the following syntax?

from pyaedt import settings
settings.enable_debug_logger=True

# your code here

this flag will make your log much more verbose and log every method you will run. that will make easier the debug but longer log.