ansys/pyaedt #1382: Bug located in .q3d.modeler.thicken_sheet

priti-ashvin-shah-ibm, 2022-07-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The direction that the polygon is drawn matters to thicken_sheet.  Counter clock wise will thicken in the direction of the thickness value using the conventional z values.  HOWEVER, if you create the poly in clockwise direction, then thickness value will become -thickness.

### Steps To Reproduce

Create_polyline two  rectangles in both clockwise and counter clock wise.  Then try to thicken_sheet with same value.  They will thicken in different directions.  

Please make it clear in the documentation as to what is expected with thicken_sheet. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

addict @ file:///home/conda/feedstock_root/build_artifacts/addict_1636818143388/work
alabaster==0.7.12
anyio==3.6.1
argon2-cffi @ file:///home/conda/feedstock_root/build_artifacts/argon2-cffi_1640817743617/work
argon2-cffi-bindings @ file:///D:/bld/argon2-cffi-bindings_1649500527917/work
astroid==2.11.5
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1618968359944/work
atomicwrites==1.4.0
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1640799537051/work
Babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1651737115240/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
backports.functools-lru-cache @ file:///home/conda/feedstock_root/build_artifacts/backports.functools_lru_cache_1618230623929/work
beautifulsoup4 @ file:///home/conda/feedstock_root/build_artifacts/beautifulsoup4_1649463573192/work
bleach @ file:///home/conda/feedstock_root/build_artifacts/bleach_1649361991009/work
branca @ file:///home/conda/feedstock_root/build_artifacts/branca_1650311966802/work
brotlipy @ file:///D:/bld/brotlipy_1648854347500/work
cached-property @ file:///home/conda/feedstock_root/build_artifacts/cached_property_1615209429212/work
certifi==2022.5.18.1
cffi @ file:///D:/bld/cffi_1636046270630/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1644853463426/work
click @ file:///D:/bld/click_1651215332191/work
click-plugins==1.1.1
cligj @ file:///home/conda/feedstock_root/build_artifacts/cligj_1633637764473/work
colorama @ file:///home/conda/feedstock_root/build_artifacts/colorama_1602866480661/work
cryptography @ file:///D:/bld/cryptography_1652967200171/work
cycler @ file:///home/conda/feedstock_root/build_artifacts/cycler_1635519461629/work
Cython @ file:///D:/bld/cython_1652809486376/work
debugpy @ file:///D:/bld/debugpy_1649586532851/work
decorator @ file:///home/conda/feedstock_root/build_artifacts/decorator_1641555617451/work
defusedxml @ file:///home/conda/feedstock_root/build_artifacts/defusedxml_1615232257335/work
descartes==1.1.0
dill==0.3.5.1
docutils @ file:///D:/bld/docutils_1648737820009/work
entrypoints @ file:///home/conda/feedstock_root/build_artifacts/entrypoints_1643888246732/work
executing @ file:///home/conda/feedstock_root/build_artifacts/executing_1646044401614/work
fastjsonschema @ file:///home/conda/feedstock_root/build_artifacts/python-fastjsonschema_1641751198313/work/dist
Fiona==1.8.21
flit_core @ file:///home/conda/feedstock_root/build_artifacts/flit-core_1645629044586/work/source/flit_core
folium @ file:///home/conda/feedstock_root/build_artifacts/folium_1638560701663/work
fonttools @ file:///D:/bld/fonttools_1651018175663/work
GDAL==3.5.0
gdspy @ file:///D:/bld/gdspy_1649850438341/work
geopandas @ file:///home/conda/feedstock_root/build_artifacts/geopandas_1638382567988/work
h5py @ file:///D:/bld/h5py_1637965026596/work
idna @ file:///home/conda/feedstock_root/build_artifacts/idna_1642433548627/work
[... 108 lines cut ...]
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml @ file:///home/conda/feedstock_root/build_artifacts/sphinxcontrib-serializinghtml_1649380998999/work
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1644872665635/work
statsmodels @ file:///D:/bld/statsmodels_1644535944524/work
sympy @ file:///D:/bld/sympy_1647802833311/work
terminado @ file:///D:/bld/terminado_1652790796353/work
threadpoolctl @ file:///home/conda/feedstock_root/build_artifacts/threadpoolctl_1643647933166/work
tinycss2 @ file:///home/conda/feedstock_root/build_artifacts/tinycss2_1637612658783/work
tomli==2.0.1
tornado @ file:///D:/bld/tornado_1648827438424/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1649051611147/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1652735690480/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1650370875435/work
unicodedata2 @ file:///D:/bld/unicodedata2_1649112137711/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1647489083693/work
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1600965781394/work
webencodings==0.5.1
websocket-client==1.3.2
widgetsnbextension @ file:///D:/bld/widgetsnbextension_1647447176344/work
win-inet-pton @ file:///D:/bld/win_inet_pton_1648771897787/work
wrapt==1.14.1
xarray @ file:///home/conda/feedstock_root/build_artifacts/xarray_1646254587173/work
xyzservices @ file:///home/conda/feedstock_root/build_artifacts/xyzservices_1649934406482/work
yapf==0.32.0
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1649012893348/work

Samuelopez-ansys, 2022-07-14:
Hi @priti-ashvin-shah-ibm ,

Could you share a small piece of code to reproduce the issue?

Thanks!

maxcapodi78, 2022-07-19:
@priti-ashvin-shah-ibm you are correct. that is a behaviour of the native AEDT API and we cannot much. I'll investigate if I can find a trick to understand the thicken direction

priti-ashvin-shah-ibm, 2022-07-21:
Let's see if we can get a working solution that I can use.  Since Qiskit Metal is not determining the polyline, I could determine the normal each time there is a polyline.

How about:
Is there a way for pyaedt to tell user the direction of normal for a polyline?   Seems like the information may already be stored in pyaedt, could you provide method to access it? 

or 
Make a subclass  method to create_polyline with an absolute z value to thicken to, ask opposed to a thickness value?

or 

Make a new method create_custom_3d_shape

maxcapodi78, 2022-07-29:
@priti-ashvin-shah-ibm  you have a property in faces that gives you the normal. you could use that one to identify the direction and then use thicken sheet accordingly.

```
>>> 
m3d
<Maxwell3d object at 0x0000000000000089>
>>> m3d.modeler.object_names
['Polyline1', 'Polyline2']
>>> m3d.modeler["Polyline1"].faces[0].normal
[0.0, -0.0, -1.0]
>>> m3d.modeler["Polyline2"].faces[0].normal
[-0.0, 0.0, 1.0]
>>> 
```

priti-ashvin-shah-ibm, 2022-07-29:
That is helpful, thank you!  For future pyaedt community, someone may want to add that information to the documentation for thicken_sheet.

closed 2022-07-29 by priti-ashvin-shah-ibm