ansys/pyaedt #1388: A second bug located in AdvancedPostProcessing.PostProcessor.create_fieldplot_surface()

obrienpja, 2022-07-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am hitting an exception in AdvancedPostProcessing.PostProcessor.create_fieldplot_surface() that I was not hitting in a previous version (0.4.83). The bug I am seeing occurs on the following line in solutions.py (line 2298 in version 0.5.1):
oname = self._postprocessor.modeler.oeditor.GetObjectNameByFaceID(index)

Here is the exception:
```self._postprocessor.modeler.oeditor.GetObjectNameByFaceID(index)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\pob_qiskit_metal_pyaedt2\lib\site-packages\pyaedt\modules\solutions.py", line 2298, in plotGeomInfo
    oname = self._postprocessor.modeler.oeditor.GetObjectNameByFaceID(index)
  File "<COMObject SetActiveEditor>", line 3, in GetObjectNameByFaceID
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024381), None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<COMObject SetActiveEditor>", line 3, in GetObjectNameByFaceID
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024381), None)
```

I am quite sure that this bug was introduced on July 1st in this commit:
https://github.com/pyansys/pyaedt/commit/d94ac54d589a740840c07551b12f0d0e359c8d37
where this line of code was first introduced.

### Steps To Reproduce

1. Instantiate an instance of Hfss() that has an Eigenmode setup
2. Analyze that setup 
3. Then run the post processor's create_fieldplot_surface() method using a syntax like this:
postProcessor.create_fieldplot_surface(objlist=[object_id], quantityName='Mag_E'), where object_id is the id of the object in your design for which you would like to plot the magnitude of the E field.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

addict @ file:///home/conda/feedstock_root/build_artifacts/addict_1636818143388/work
aiohttp==3.8.1
aiosignal==1.2.0
alabaster==0.7.12
anyio==3.6.1
appdirs==1.4.4
argon2-cffi @ file:///home/conda/feedstock_root/build_artifacts/argon2-cffi_1640817743617/work
argon2-cffi-bindings @ file:///D:/bld/argon2-cffi-bindings_1649500527917/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1618968359944/work
async-timeout==4.0.2
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1640799537051/work
Babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1655419414885/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
backports.functools-lru-cache @ file:///home/conda/feedstock_root/build_artifacts/backports.functools_lru_cache_1618230623929/work
beautifulsoup4 @ file:///home/conda/feedstock_root/build_artifacts/beautifulsoup4_1649463573192/work
bleach @ file:///home/conda/feedstock_root/build_artifacts/bleach_1656355450470/work
branca @ file:///home/conda/feedstock_root/build_artifacts/branca_1650311966802/work
brotlipy @ file:///D:/bld/brotlipy_1648854347500/work
cached-property @ file:///home/conda/feedstock_root/build_artifacts/cached_property_1615209429212/work
certifi==2022.6.15
cffi @ file:///D:/bld/cffi_1656782940775/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1655906222726/work
click @ file:///D:/bld/click_1651215332191/work
[... 127 lines cut ...]
sphinxcontrib-serializinghtml @ file:///home/conda/feedstock_root/build_artifacts/sphinxcontrib-serializinghtml_1649380998999/work
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1655315839047/work
statsmodels @ file:///D:/bld/statsmodels_1644535944524/work
sympy @ file:///D:/bld/sympy_1647802833311/work
terminado @ file:///D:/bld/terminado_1652790796353/work
threadpoolctl @ file:///home/conda/feedstock_root/build_artifacts/threadpoolctl_1643647933166/work
tinycss2 @ file:///home/conda/feedstock_root/build_artifacts/tinycss2_1637612658783/work
toml @ file:///home/conda/feedstock_root/build_artifacts/toml_1604308577558/work
tornado @ file:///D:/bld/tornado_1656937966227/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1649051611147/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1655411388954/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1656706066251/work
unicodedata2 @ file:///D:/bld/unicodedata2_1649112137711/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1657224465922/work
vtk==9.2.0rc1
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1600965781394/work
webencodings==0.5.1
websocket-client==1.3.3
widgetsnbextension @ file:///home/conda/feedstock_root/build_artifacts/widgetsnbextension_1655939017940/work
win-inet-pton @ file:///D:/bld/win_inet_pton_1648771897787/work
wslink==1.6.6
xarray @ file:///home/conda/feedstock_root/build_artifacts/xarray_1646254587173/work
xyzservices @ file:///home/conda/feedstock_root/build_artifacts/xyzservices_1655861233352/work
yarl==1.7.2
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1649012893348/work

maxcapodi78, 2022-07-19:
@obrienpja you are right. I've introduced a regression by adding the possibility to port surface fields on on models. 
I've fixed in #1423

obrienpja, 2022-07-19:
Excellent, thank you for the fix!

closed 2022-07-19 by obrienpja