ansys/pyaedt #1402: Add ExportNetworkData in Q3d.post

linmingchih, 2022-07-17:
### Description of the feature

I can't find the network and equivalent circuit export in Q3D.post:

[image]

Samuelopez-ansys, 2022-09-01:
Hi @gmalinve ,

I think you implemented this feature recently right?

Please if I am right, could you specify where is the method to @linmingchih and close the issue?

Thanks!

gmalinve, 2022-09-01:
Hi @Samuelopez-ansys,

I implemented the ExportMatrixData to export RLCG matrix. I didn't do anything about the export of the equivalent circuit, I shuold check.

gmalinve, 2022-09-01:
Hi @Samuelopez-ansys , @linmingchih ,

so the export RLCG matrix data has been implemented in a previous PR (#1565 ) but the export equivalent circuit no. 
It's a completely different feature. 
Should we open a new issue to implement the export equivalent circuit and close this one?

gmalinve, 2022-09-01:
@linmingchih 
we keep this issue to develop the export equivalent circuit feature. 
I'll assign to myself!

closed 2022-09-09 by gmalinve