ansys/pyaedt #1411: Refactor place_in_layout_3d_placement to support specifying target layer

isaacansys, 2022-07-18:
### Description of the feature

Currently [place_in_layout_3d_placement](https://github.com/pyansys/pyaedt/blob/231b8493bf5a82bc3ee43e8a3abb47005910ee99/pyaedt/edb_core/stackup.py#L593) will only place a 3D Component on either the top or bottom signal layer of a design.  For some stackups, such as when a shielding layer is present, this requires resetting the Z-location of the placed component after placement with this function to place the component on a desired "inner" signal layer.

### Steps for implementing the feature

1. Implement a new function that places a 3D Component above or below a layer specified by name
2. Refactor [place_in_layout_3d_placement](https://github.com/pyansys/pyaedt/blob/231b8493bf5a82bc3ee43e8a3abb47005910ee99/pyaedt/edb_core/stackup.py#L593) to use the new function, passing either the top or bottom signal layer

svandenb-dev, 2023-02-20:
@isaacansys can we close this issue ?

maxcapodi78, 2023-04-17:
@svandenb-dev I think this is still a request needed. I've recently implemented a new method place_instance that is similar to place_in_layout_3d_placement but gives more flexibility especially when we need to place more than 1 instance in the same layout. maybe the refactor can affect this method too