ansys/pyaedt #1437: Cannot install test requirements with CPython 3.7

isaacansys, 2022-07-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The requirements_test.txt file seems to contain some packages that are duplicates of the install-requires but which are fixed at versions beyond what is available for CPython 3.7 on Linux.  This results in not being able to install the requirements with this file.

### Steps To Reproduce

```
git clone https://github.com/pyansys/pyaedt.git
cd pyaedt
export ANSYSEM_ROOT=/path/to/AnsysEM/v222/Linux64
export LD_LIBRARY_PATH=$ANSYSEM_ROOT222/common/mono/Linux64/lib:$ANSYSEM_ROOT222/Delcross
$ANSYSEM_ROOT222/commonfiles/CPython/3_7/linx64/Release/python/bin/python3.7 -m venv venv
. venv/bin/activate
pip install .
pip install -r requirements_test.txt
```

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.7

### Installed packages

None since reproducing involves setting up a virtual environment from scratch.

MaxJPRey, 2022-08-02:
@isaacansys Can you share the error message that you get when trying to install the dependencies?
Thanks

isaacansys, 2022-08-03:
```
ERROR: No matching distribution found for numpy==1.23.1
```

MaxJPRey, 2022-08-03:
@isaacansys Thanks, I will try to fix those dependencies issue for python 3.7.

PR #1506 (2022-08-08): Numpy stopped supporting Python 3.7 since release 1.22.0.
Numpy is not supporting Python 3.7 starting with release 1.22.0
See: https://numpy.org/devdocs/release/1.22.0-notes.html

Fix #1437 .
closed 2022-08-08 by MaxJPRey