ansys/pyaedt #1439: Bug located in ... https://aedtdocs.pyansys.com/API/Constants.html

PipKat, 2022-07-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The images in https://aedtdocs.pyansys.com/API/Constants.html were captured in dark mode and aren't readable. All images should be captured in light mode.[](url)

### Steps To Reproduce

Go to HTML page in PyAEDT doc: https://aedtdocs.pyansys.com/API/Constants.html

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.12
Babel==2.10.3
certifi==2022.6.15
charset-normalizer==2.0.12
colorama==0.4.5
distlib==0.3.5
docutils==0.18.1
filelock==3.7.1
idna==3.3
imagesize==1.3.0
Jinja2==3.1.2
MarkupSafe==2.1.1
packaging==21.3
platformdirs==2.5.2
Pygments==2.12.0
pyparsing==3.0.9
pytz==2022.1
requests==2.28.0
six==1.16.0
snowballstemmer==2.2.0
Sphinx==5.0.2
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
urllib3==1.26.9
virtualenv==20.15.1

fwlfHS, 2022-07-21:
You mean the deprecation warnings? These are formated with https://aedtdocs.pyansys.com/_static/css/ansys_sphinx_theme.css
Text color: --pst-color-text-base (to dark in light mode)
Background color: --pst-color-danger (still black in light mode)

PipKat, 2022-07-22:
I understand that the formatting is done through the stylesheet. However, look at the following renderings in the light and dark modes. I thought there was guidance that the screen captures themselves should be taken while in the light mode. However, I can't find that guidance in the PyAnsys Developer's Guide. There is discussion on how to best handle this situation with the PyAnsys team. I'm copying key members here so they can chime in. It seems the PyAnsys Developer's guide should include how best to capture a screen for inclusion in an image at the very end of the "The doc/ directory" section in this [topic](https://dev.docs.pyansys.com/packaging/structure.html). @jorgepiloto @RobPasMue 

[image]
[image]

fwlfHS, 2022-07-22:
Hi Kathy, then there are two topics:

1. Guide line on screen capturing
2. fix the stylesheet (because deprecation warnings are text not images)

PipKat, 2022-07-22:
Excellent point. I didn't realize the deprecation warnings were text! Based on how they looked, I assumed that they were images. That was stupid of me. Thanks!

MaxJPRey, 2022-08-02:
This has been fixed recently in: 
https://github.com/ansys/ansys-sphinx-theme/pull/85

Here is the new rendering.
[image]

closed 2022-08-02 by MaxJPRey