ansys/pyaedt #1447: Add "Syntax Request" label for new issues

obrienpja, 2022-07-25:
### Description of the feature

I was wondering if it would useful to have a "syntax request" label for new issues. Sometimes it seems like it should be possible for me to use pyaedt as is, but I can't figure out the syntax from the API documentation. In these cases, I could choose the "syntax request" label as opposed to the "bug" label. Just an idea. Thanks!

### Steps for implementing the feature

Create "syntax request" label for new issues (or something like that).

MaxJPRey, 2022-07-26:
@maxcapodi78 I could add a label for "help/support".

obrienpja, 2022-07-26:
Perfect, thank you!

maxcapodi78, 2022-07-29:
@MaxJPRey Can you take this action? I think it is a good idea to add it to issue templates

MaxJPRey, 2022-08-01:
@maxcapodi78 Yes, I will add this label.

PR #2208 (2022-12-27): Add new issue template for the enhancements of the API design.
Fix #1447 .
closed 2022-12-27 by MaxJPRey