ansys/pyaedt #1453: Read/export R,L matrix in Maxwell3D after solution

e-rando, 2022-07-25:
### Description of the feature

A function that lets you read/export R,L matrix in Maxwell3D after solution.

maxcapodi78, 2022-07-29:
@e-rando for which Maxwell Solution Type? Should be also applied to Maxwell2D? Can you share a small example?
@gmalinve can you take it?

e-rando, 2022-07-29:
@gmalinve @maxcapodi78  I only use Maxwell3D with EddyCurrent, but I guess it is interesting for any of them. 
What I meant to read is this matrix (setup -> solutions -> Matrix):

[image]

For my application, it is very relevant to be able to access not only to 'Reduce matrix: original', but also to the matrices generated with the matrix reduction (join in parallel/series)

In my opinion, the most useful way to acquire this data is with a list or a list of lists that contain the parameters. Probably two different outputs, one for R and one for L would be best.

gmalinve, 2022-08-03:
Hi @e-rando , would you have a ready-made example so I can understand better what you intend to do please? 
Maybe an example with the reduced matrix and the ones generated with matrix reduction.
thanks!

gmalinve, 2022-08-08:
Hi @e-rando . I have implemented the requested feature for Maxwell 2D and 3D, PR is created and waiting for approval!

closed 2022-08-09 by Samuelopez-ansys