ansys/pyaedt #1461: edb.py: import_layout_pcb has a filehandler bug

Jan-Hammer, 2022-07-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When calling edb.import_cadence_file() and subsequent edb.import_layout_pcb() via pyadet the ansystranslator file on Windows is called with the wrong arguments causing the translation to exit with error. 

### Steps To Reproduce

## Given that the PCB cadence file is named "Board.brd"
import os, subprocess
from pyaedt import edb
app=Edb()
dir=os.getcwd()
Board_name=os.path.join(dir,"Board.brd")
app.import_cadence_file(Board_Name)

##error msg: "Translator failed to translate"

Malfunctioning code: 
edb.py line 520-525: 
            cmd_translator = [
                command,
                input_file,
                os.path.join(working_dir, aedb_name),
                '-l="{}"'.format(os.path.join(working_dir, "Translator.log")),
            ]
Fix:
replace  '-l="{}"'.format(os.path.join(working_dir, "Translator.log")) 
with  "-l='{}'".format(os.path.join(working_dir, "Translator.log"))

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp @ file:///C:/ci/aiohttp_1646806572557/work
aiosignal @ file:///tmp/build/80754af9/aiosignal_1637843061372/work
alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
anaconda-client @ file:///C:/ci/anaconda-client_1635342725944/work
anaconda-navigator==2.2.0
anaconda-project @ file:///tmp/build/80754af9/anaconda-project_1637161053845/work
anyio @ file:///C:/ci/anyio_1644481921011/work/dist
appdirs==1.4.4
argon2-cffi @ file:///opt/conda/conda-bld/argon2-cffi_1645000214183/work
argon2-cffi-bindings @ file:///C:/ci/argon2-cffi-bindings_1644551690056/work
arrow @ file:///opt/conda/conda-bld/arrow_1649166651673/work
astroid @ file:///C:/ci/astroid_1628063282661/work
astropy @ file:///C:/ci/astropy_1650634291321/work
asttokens @ file:///opt/conda/conda-bld/asttokens_1646925590279/work
async-timeout @ file:///tmp/build/80754af9/async-timeout_1637851218186/work
atomicwrites==1.4.0
attrs @ file:///opt/conda/conda-bld/attrs_1642510447205/work
Automat @ file:///tmp/build/80754af9/automat_1600298431173/work
autopep8 @ file:///opt/conda/conda-bld/autopep8_1639166893812/work
Babel @ file:///tmp/build/80754af9/babel_1620871417480/work
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
backports.functools-lru-cache @ file:///tmp/build/80754af9/backports.functools_lru_cache_1618170165463/work
backports.tempfile @ file:///home/linux1/recipes/ci/backports.tempfile_1610991236607/work
backports.weakref==1.0.post1
bcrypt @ file:///C:/ci/bcrypt_1607022693089/work
beautifulsoup4 @ file:///C:/ci/beautifulsoup4_1650293025093/work
binaryornot @ file:///tmp/build/80754af9/binaryornot_1617751525010/work
[... 286 lines cut ...]
ujson @ file:///C:/ci/ujson_1648044223886/work
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3 @ file:///C:/ci/urllib3_1650639883891/work
vtk==9.1.0
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci/watchdog_1638367441841/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci/websocket-client_1614804375980/work
Werkzeug @ file:///opt/conda/conda-bld/werkzeug_1645628268370/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1644991377168/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/ci/wrapt_1607574570428/work
wslink==1.6.6
xarray @ file:///opt/conda/conda-bld/xarray_1639166117697/work
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///opt/conda/conda-bld/xlsxwriter_1649073856329/work
xlwings==0.24.9
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl @ file:///C:/ci/yarl_1606940155993/work
zict==2.0.0
zipp @ file:///opt/conda/conda-bld/zipp_1641824620731/work
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

maxcapodi78, 2022-07-29:
fixed in #1482

closed 2022-07-29 by maxcapodi78

maxcapodi78, 2022-07-29:
@Jan-Hammer one thing I forgot. you may directly do:

```
import os, subprocess
from pyaedt import edb
dir=os.getcwd()
Board_name=os.path.join(dir,"Board.brd")
app=Edb(Board_Name)
```

Edb will take care of conversion and return the edb object.