ansys/pyaedt #1464: Edb.export_maxwell net_list parameter does not result in exporting of selected nets

Jan-Hammer, 2022-07-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When using Edb.export_maxwell() with the net_list parameter, the software still exports all nets although nets are passed correctly to the generated python file (export_cad.py). 

Expected behavior would be that only the nets in 'net_list' would be exported 

### Steps To Reproduce

from pyaedt import Edb 
import os 
path=os.path.getcdw()
ProjectPath=os.path.join(path,'ProjectName')
app=Edb(ProjectPath)
list=['Net1','Net2']
app.export_maxwell(path_to_output=ProjectPath,net_list=list,aedt_file_name="Output")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp @ file:///C:/ci/aiohttp_1646806572557/work
aiosignal @ file:///tmp/build/80754af9/aiosignal_1637843061372/work
alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
anaconda-client @ file:///C:/ci/anaconda-client_1635342725944/work
anaconda-navigator==2.2.0
anaconda-project @ file:///tmp/build/80754af9/anaconda-project_1637161053845/work
anyio @ file:///C:/ci/anyio_1644481921011/work/dist
appdirs==1.4.4
argon2-cffi @ file:///opt/conda/conda-bld/argon2-cffi_1645000214183/work
argon2-cffi-bindings @ file:///C:/ci/argon2-cffi-bindings_1644551690056/work
arrow @ file:///opt/conda/conda-bld/arrow_1649166651673/work
astroid @ file:///C:/ci/astroid_1628063282661/work
astropy @ file:///C:/ci/astropy_1650634291321/work
asttokens @ file:///opt/conda/conda-bld/asttokens_1646925590279/work
async-timeout @ file:///tmp/build/80754af9/async-timeout_1637851218186/work
atomicwrites==1.4.0
attrs @ file:///opt/conda/conda-bld/attrs_1642510447205/work
Automat @ file:///tmp/build/80754af9/automat_1600298431173/work
autopep8 @ file:///opt/conda/conda-bld/autopep8_1639166893812/work
Babel @ file:///tmp/build/80754af9/babel_1620871417480/work
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
backports.functools-lru-cache @ file:///tmp/build/80754af9/backports.functools_lru_cache_1618170165463/work
backports.tempfile @ file:///home/linux1/recipes/ci/backports.tempfile_1610991236607/work
backports.weakref==1.0.post1
bcrypt @ file:///C:/ci/bcrypt_1607022693089/work
beautifulsoup4 @ file:///C:/ci/beautifulsoup4_1650293025093/work
binaryornot @ file:///tmp/build/80754af9/binaryornot_1617751525010/work
bitarray @ file:///C:/ci/bitarray_1648739663053/work
bkcharts==0.2
black==19.10b0
bleach @ file:///opt/conda/conda-bld/bleach_1641577558959/work
bokeh @ file:///C:/ci/bokeh_1638362966927/work
boto3 @ file:///opt/conda/conda-bld/boto3_1649078879353/work
botocore @ file:///opt/conda/conda-bld/botocore_1649076662316/work
Bottleneck @ file:///C:/ci/bottleneck_1648010904582/work
brotlipy==0.7.0
cachetools @ file:///tmp/build/80754af9/cachetools_1619597386817/work
certifi==2021.10.8
cffi @ file:///C:/ci_310/cffi_1642682485096/work
[... 274 lines cut ...]
ujson @ file:///C:/ci/ujson_1648044223886/work
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3 @ file:///C:/ci/urllib3_1650639883891/work
vtk==9.1.0
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci/watchdog_1638367441841/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci/websocket-client_1614804375980/work
Werkzeug @ file:///opt/conda/conda-bld/werkzeug_1645628268370/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1644991377168/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/ci/wrapt_1607574570428/work
wslink==1.6.6
xarray @ file:///opt/conda/conda-bld/xarray_1639166117697/work
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///opt/conda/conda-bld/xlsxwriter_1649073856329/work
xlwings==0.24.9
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl @ file:///C:/ci/yarl_1606940155993/work
zict==2.0.0
zipp @ file:///opt/conda/conda-bld/zipp_1641824620731/work
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

maxcapodi78, 2022-07-28:
@Jan-Hammer the api exports the selected nets plus all the power/ground nets. those net infacts, cannot be excluded by Siwave from the export. An option could be to remove them and then readd but I've to check feasibility

closed 2022-07-28 by maxcapodi78

Jan-Hammer, 2022-07-28:
@maxcapodi78 If that is the case than the documentation of export_maxwell needs to be adjusted 

"net_list[list], optional
List of nets to export only if certain ones are to be exported. The default is None, in which case all nets are exported."

Jan-Hammer, 2022-07-28:
@maxcapodi78  Removing Nets prior export is a functional work-around! Thanks for pointing this out.