ansys/pyaedt #1467: Bug located in Modeler2D: object has no attribute 'components_3d_names' (in duplicate_and_mirror)

romainlejeune, 2022-07-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

when using Modeler2D.duplicate_and_mirror() I have the following error.

I guess it's because of the changes #1385 where 
def **components_3d_names**(self) was renamed into def user_defined_component_names(self) in Primitives.py
but not in line 2431 of Modeler.py in def duplicate_and_mirror() 
        if is_3d_comp:
            orig_3d = [i for i in self.**components_3d_names**]

by the way, comparing files seems to show that the change seems to be done for def duplicate_around_axis()
~~~~~~~~~~~~~~~ERROR ~~~~~~~~~~~~

pyaedt error: Method duplicate_and_mirror Failed:  Attribute Error. Please Check again
Method duplicate_and_mirror Failed:  Attribute Error. Please Check again
pyaedt error: 'Modeler2D' object has no attribute '**components_3d_names**'
'Modeler2D' object has no attribute 'components_3d_names'
pyaedt error: Arguments Provided:
Arguments Provided:
pyaedt error:     objid = [<pyaedt.modeler.Primitives.Polyline object at 0x000002193745E648>, <pyaedt.modeler.Primitives.Polyline object at 0x000002193745E208>]
    objid = [<pyaedt.modeler.Primitives.Polyline object at 0x000002193745E648>, <pyaedt.modeler.Primitives.Polyline object at 0x000002193745E208>]
pyaedt error:     position = [0, 0, 0]
    position = [0, 0, 0]
pyaedt error:     vector = ['cos((360deg/SymmetryFactor/2)+90deg)', 'sin((360deg/SymmetryFactor/2)+90deg)', 0]
    vector = ['cos((360deg/SymmetryFactor/2)+90deg)', 'sin((360deg/SymmetryFactor/2)+90deg)', 0]
pyaedt error:     is_3d_comp = True
    is_3d_comp = True
pyaedt error:     duplicate_assignment = None
    duplicate_assignment = None
pyaedt error: Error in :   File "C:\Users\Sharath\Desktop\Solution-Applications\solutions-app-webapp-poc\venv\lib\site-packages\pyaedt\modeler\Modeler.py", line 2431, in duplicate_and_mirror
Error in :   File "C:\Users\Sharath\Desktop\Solution-Applications\solutions-app-webapp-poc\venv\lib\site-packages\pyaedt\modeler\Modeler.py", line 2431, in duplicate_and_mirror
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=duplicate_and_mirror
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=duplicate_and_mirror
pyaedt error: Method duplicate_and_mirror Failed:  Attribute Error. Please Check again
Method duplicate_and_mirror Failed:  Attribute Error. Please Check again
pyaedt error: 'Modeler2D' object has no attribute 'components_3d_names'
'Modeler2D' object has no attribute 'components_3d_names'
pyaedt error: Arguments Provided:
Arguments Provided:
pyaedt error:     objid = [<pyaedt.modeler.Primitives.Polyline object at 0x000002193A852CC8>, <pyaedt.modeler.Primitives.Polyline object at 0x0000021939813708>]
    objid = [<pyaedt.modeler.Primitives.Polyline object at 0x000002193A852CC8>, <pyaedt.modeler.Primitives.Polyline object at 0x0000021939813708>]
pyaedt error:     position = [0, 0, 0]
    position = [0, 0, 0]
pyaedt error:     vector = ['cos((360deg/SymmetryFactor/2)+90deg)', 'sin((360deg/SymmetryFactor/2)+90deg)', 0]
    vector = ['cos((360deg/SymmetryFactor/2)+90deg)', 'sin((360deg/SymmetryFactor/2)+90deg)', 0]
pyaedt error:     is_3d_comp = True
    is_3d_comp = True
pyaedt error:     duplicate_assignment = None
    duplicate_assignment = None
pyaedt error: Error in :   File "C:\Users\Sharath\Desktop\Solution-Applications\solutions-app-webapp-poc\venv\lib\site-packages\pyaedt\modeler\Modeler.py", line 2431, in duplicate_and_mirror
Error in :   File "C:\Users\Sharath\Desktop\Solution-Applications\solutions-app-webapp-poc\venv\lib\site-packages\pyaedt\modeler\Modeler.py", line 2431, in duplicate_and_mirror
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=duplicate_and_mirror
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=duplicate_and_mirror
2022-07-27 15:39:43.135 (  39.582s) [                ]vtkWin32OpenGLRenderWin:680    ERR| vtkWin32OpenGLRenderWindow (0000021933393060): failed to get wglChoosePixelFormatARB
ERROR:root:failed to get wglChoosePixelFormatARB
2022-07-27 15:39:43.137 (  39.583s) [                ]vtkWin32OpenGLRenderWin:758    ERR| vtkWin32OpenGLRenderWindow (0000021933393060): failed to get valid pixel format.
ERROR:root:failed to get valid pixel format.
2022-07-27 15:39:43.137 (  39.584s) [                ]vtkOpenGLRenderWindow.c:493    ERR| vtkWin32OpenGLRenderWindow (0000021933393060): GLEW could not be initialized: Missing GL version
ERROR:root:GLEW could not be initialized: Missing GL version

### Steps To Reproduce

You can run the code here: https://aedtdocs.pyansys.com/examples/02-Maxwell/Maxwell2D_NissanLeaf.html#sphx-glr-examples-02-maxwell-maxwell2d-nissanleaf-py

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

appdirs==1.4.4
async-timeout==4.0.2
attrs==21.4.0
Babel==2.10.3
backcall==0.2.0
certifi==2022.6.15
charset-normalizer==2.0.12
colorama==0.4.5
cycler==0.11.0
debugpy==1.6.0
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
executing==0.8.3
fastjsonschema==2.15.3
fonttools==4.33.3
frozenlist==1.3.0
idna==3.3
ipython-genutils==0.2.0
json5==0.9.8
jupyterlab-pygments==0.2.2
kiwisolver==1.4.3
MarkupSafe==2.1.1
mistune==0.8.4
multidict==6.0.2
nest-asyncio==1.5.5
numpy==1.23.0
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.1.1
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.29
psutil==5.9.1
pure-eval==0.2.2
pycparser==2.21
Pygments==2.12.0
pyparsing==3.0.9
pyrsistent==0.18.1
pytz==2022.1
pywin32==303
pywinpty==2.0.5
pyzmq==23.2.0
requests==2.28.0
rpyc==5.0.1
scooby==0.5.12
Send2Trash==1.8.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.2.post1
terminado==0.15.0
tinycss2==1.1.1
tornado==6.1
traitlets==5.3.0
urllib3==1.26.9
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.3
yarl==1.7.2
zipp==3.8.0

maxcapodi78, 2022-07-28:
Hi @romainlejeune  I've fixed it

romainlejeune, 2022-07-28:
Hi @maxcapodi78 , thanks a lot!

PR #1472 (2022-07-28): Fix 1467
Fix #1467
closed 2022-07-28 by maxcapodi78