ansys/pyaedt #1478: export edb to maxwell - Vias are exported filled regardless of plating thickness

Jan-Hammer, 2022-07-28:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

plating thickness of padstacks are adjustable via app.core_padstack.hole_plating_thickness. However, exporting in Maxwell results in the export of filled vias instead of the correct plating thickness. Is the default export 100% Via fill ? If yes, what is the option to disable it.

### Steps To Reproduce

from pyaedt import Edb,Maxwell3d
import os, 

path=os.getcwd()
ImportName='Import'
ExportName='Export'
PlatingThickness=0.1E-3

AEDB_file=os.path.join(path,ImportName+'.aedb')
app=Edb(AEDB_file,edbversion='2021.2')

PadStackNames=[x for x in app.core_padstack.padstacks]
for x in PadStackNames:
    try:
        app.core_padstack.padstacks[x].hole_plating_ratio=0
        app.core_padstack.padstacks[x].hole_plating_thickness=PlatingThickness
    except:
        print("Plating thickness change caused error in padstack: "+x)
app.save_edb()

options_config = {
'LAUNCH_MAXWELL' : 0,
'NUM_PADS_FACET_COUNT': 8,
'NUM_ANTIPAD_FACET_COUNT': 8,
'DEFAULT_SOLDERBALL_FACET_COUNT' : 8,
'VIA_SEGMENTS' : 8,
'DEFAULT_BONDWIRE_FACET_COUNT' : 6,
'UNITE_NETS' : 0,
'EXCLUDE_TERMINALS_FROM_UNITE' : 1,
'SIMPLY_VIA_MODEL' : 0,
'IGNORE_DIELECTRICS' : 0,
'SEPARATE_DIELECTRICS' : 1,
'UNITE_LAYERS_WITH_SAME_MATERIALS' : 1,
'IGNORE_UNCONNECTED_PADS' : 0,
'CLIP_TRACES' : 0,
'CUT_DIELECTRICS' : 0,
'CREATE_SHEET_BODIES' : 0,
'GENERATE_TERMINALS' : 0,
'IGNORE_PLANES_WITH_AREA_LESS_THAN_THRESOLD' : 1,
'IGNORE_FLOAT_BODIES' : 0,
'MIN_PLANE_AREA' : 0.358979,
'MIN_EDGE_LENGTH_PADS' : '1um',
'MIN_EDGE_LENGTH_PLANES' : '1um',
'MIN_EDGE_LENGTH_TRACES' : '1um',
'MIN_DIELECTRIC_EDGE_LENGTH' : '10um',
'DIELECTRIC_EXPANSION_FACTOR' : 0.100000,
'IGNORE_HOLES' : 0,
'MIN_HOLE_AREA' : 0.358979,
'REMOVE-PLATING_TAILS' : 0,
'SUBTRACT_METAL_FROM_SUBSTRATE' : 0,
'DISCRETIZE_ARCS' : 0,
'CHOP_TRACE_ENDS' : 0,
'AIRBOX_THICKNESS_FACTOR' : 1.100000,
'AIRBOX_PAD_AMOUNT_PLUS_Z' : 0.500000,
'AIRBOX_PAD_AMOUNT_MINUS_Z' : 0.500000,
'PORT_PAD_AMOUNT' : 0.500000,
'CREATE_PORTS_FOR_PWR_GND_NETS' : 0,
'PORTS_FOR_PWR_GND_NETS' : 0,
'LAUNCH_HFSS' : 0,
[... 340 lines cut ...]
ujson @ file:///C:/ci/ujson_1648044223886/work
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3 @ file:///C:/ci/urllib3_1650639883891/work
vtk==9.1.0
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci/watchdog_1638367441841/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci/websocket-client_1614804375980/work
Werkzeug @ file:///opt/conda/conda-bld/werkzeug_1645628268370/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1644991377168/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/ci/wrapt_1607574570428/work
wslink==1.6.6
xarray @ file:///opt/conda/conda-bld/xarray_1639166117697/work
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///opt/conda/conda-bld/xlsxwriter_1649073856329/work
xlwings==0.24.9
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl @ file:///C:/ci/yarl_1606940155993/work
zict==2.0.0
zipp @ file:///opt/conda/conda-bld/zipp_1641824620731/work
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

maxcapodi78, 2022-07-29:
@Jan-Hammer this is a current Siwave Ecad to Mcad limitation. We could fill an enhancement request and once it is there pyaedt can support it easily

maxcapodi78, 2022-07-29:
@smedikon can you follow up for enhancement request?

closed 2022-07-29 by maxcapodi78

Jan-Hammer, 2022-07-29:
@maxcapodi78 The GUI version of SiWave supports this feature. Just tested it with the .siw file that has been generated by pyaedt and I was able to export to maxwell with the correct plating thickness. 
I have a hunch that the 100% via fill option (see image) is enabled during the pyaedt export. This option seems also not documented in the online as well as the offline help of SiWave.

[image]