ansys/pyaedt #1499: Add StdErr checking when Unit tests are run

kbhagat-ansys, 2022-08-03:
### Description of the feature

When PyAEDT Unit tests are run, it would be great if it can report what is written out to StdErr and report a fail if anything is written out. pytest reports some useful actions to do for the unit tests and it gets funnelled to StdErr. 

E.g. Here in this [PR (Expand Unit Testing and then go to the end)](https://github.com/pyansys/pyaedt/runs/7562364981?check_suite_focus=true), you will notice that the pytest recommendations are written at the end(i.e. StdErr) and one has to hunt for it. 
[image]

kbhagat-ansys, 2023-10-30:
@Samuelopez-ansys  do you have the PR which fixed this?

Samuelopez-ansys, 2023-10-31:
Hi @kbhagat-ansys,

This issue was abandoned for more than 1 year, and PyAEDT unit tests changed a lot this year. Please reopen it, and if you have time implement the modification you expect. The URL you shared is not valid anymore.

Thanks