ansys/pyaedt #1518: Add ability to edit stackup layer name in EDB

skandak-ansys, 2022-08-08:
### Description of the feature

[image]

Here, `pyaedt `allows users to change all properties associated with a specific layer. However, it doesn't allow users to change layer name. 

It is a useful feature to have the ability to change the layer name.

### Steps for implementing the feature

See pic above.

skandak-ansys, 2022-08-08:
cc - @mahesh-dhanekula

Samuelopez-ansys, 2022-08-12:
@skandak-ansys is this feature already merged?

If it is, please could you close the issue?

Thank you

mahesh-dhanekula, 2022-08-12:
@Samuelopez-ansys 

Let me create the pull request for this feature, will request review from you.

pyAnsysUser, 2022-08-22:
my understanding is this functionality is included in EDBLayer.update_layer_vals() ? The documentation here is not very precise.

mahesh-dhanekula, 2022-08-22:
@pyAnsysUser 
Except for the name everything can be modified in layer attributes

pyAnsysUser, 2022-08-26:
This might be the right way to go, but #1064 may point to a logical bug located in the layer properties when adding a layer. I am still working my way through the code that sets the new id value for the added layer.

SHF101202021, 2022-08-26:
@pyAnsysUser 
Thanks for you replies
After figuring out some solutions for #1064,  I think the proposed method here would not happen like you thought

mahesh-dhanekula, 2022-09-05:
@pyAnsysUser 
Seems like this solution is failing on my end if I tried to modify the layer name second time. Can you confirm by verifying this behavior ?

Samuelopez-ansys, 2022-09-05:
@svandenb-dev please could you check this issue? Thanks!

ring630, 2022-09-21:
@Samuelopez-ansys @SHF101202021 @mahesh-dhanekula @pyAnsysUser @skandak-ansys please use the new stackup method to change layer name.

`edbapp.stackup["TOP"].name = "new_name"`

Below is another example
https://github.com/pyansys/pyaedt/blob/7707a8875584270ed1c304e80be416b29d45f4b4/_unittest/test_00_EDB.py#L2036-L2046

closed 2022-09-22 by ring630