ansys/pyaedt #1526: [PostProsessing] specify a time period for report

mehs1006, 2022-08-10:
Hi,
I wonder how to specify a time period by PyAEDT when creating a report.
I used below dictionary as “variation input” but got the all time period of "$Is_rms=4A", so it doesn't work for time interval.

_ Dict_variations={"Time": ["0.4001s","0.5s"], "$Is_rms": ["4A"]} _

Please help me.
Thank you.

BR,
/Meng-Ju

maxcapodi78, 2022-08-31:
Hello @mehs1006 can you provide a full example on what you want to do? not sure on which tool/solver you want it

Samuelopez-ansys, 2022-09-27:
Hi @mehs1006 , please could you provide a full example to reproduce the issue? 

We will close the issue in 1 week.

Thanks

mehs1006, 2022-09-27:
Hi,
Sorry for the late reply. It is in the **Maxwell2D.post.create_report** module.
Here's the code.

`from pyaedt import Maxwell2d
m2d = Maxwell2d(
    projectname = "IM_15kW_PI_LT2NL", 
    designname = "Crnt_CLL_VTS_NL",
    );
ExportVars = [
    "InputCurrent(PhaseA)","InputCurrent(PhaseB)","InputCurrent(PhaseC)",
    "FluxLinkage(PhaseA)","FluxLinkage(PhaseB)","FluxLinkage(PhaseC)", 
    "InducedVoltage(PhaseA)","InducedVoltage(PhaseB)","InducedVoltage(PhaseC)"
    ];
for I in range(1,10):
    NameTemp = "DT_Is_rms_NL_"+str(int(I))+"A";
    Dict_variations={"Time": ["0.4001s","0.5s"], "$Is_rms": [str(int(I))+"A"]};
    m2d.post.create_report(
        ExportVars, report_category="Transient", plot_type="data table", 
        domain="Sweep", variations=Dict_variations, 
        primary_sweep_variable="Time",
        plotname=NameTemp
        );
for I in range(1,10):
    NameTemp = "DT_Is_rms_NL_"+str(int(I))+"A";
    m2d.post.export_report_to_file(
        "C:/Users/mengju/Box/PhDProject/00_ICEM2022/03Analysis/New/03RawData/NLT", NameTemp, ".txt");`

The output .txt file has the report of "$Is_rms=4A" for Time= ["0s", "0.5s"].
But what I want is only Time= ["0.4001s", "0.5s"].

Please help me.
Thank you.

BR,
/Meng-Ju

Samuelopez-ansys, 2022-11-11:
Hi @mehs1006 ,

Sorry for the late reply, please could you share with us the AEDT model to run this code snippet.

Please send it to pyansys.support@ansys.com

Best regards,
Samuel

gmalinve, 2022-12-20:
Hi @mehs1006 , please could you provide a full example to reproduce the issue?

I'll close the issue otherwise.

Thanks!

closed 2023-02-13 by Samuelopez-ansys