ansys/pyaedt #1546: Bug located in hfss.export_3d_model()

wdmaiansys, 2022-08-16:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to export the 3d model in HFSS using:hfss.export_3d_model('model','E:',fileFormat='.step', object_list=None,removed_objects=None)

However, it gives error message as: pyaedt error: Error in :   File "c:\Python39\lib\site-packages\pyaedt\application\Analysis3D.py", line 530, in export_3d_model

If I use hfss.export_3d_model('model','E:',fileFormat='.stl', object_list=None,removed_objects=None), It would return true and generate a possibly broken file that cannot be opened by Windows 3D viewer. It is strange as I know HFSS does not support stl export.

### Steps To Reproduce

example:

f=*/project name

non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "True").lower() in ("true", "1", "t")
desktopVersion = "2023.1"
NewThread = False
desktop = Desktop(desktopVersion, non_graphical=non_graphical, new_desktop_session=NewThread)

remove_project_lock(f)
hfss = Hfss(f)
hfss.export_3d_model('model','E:',fileFormat='.step', object_list=None,removed_objects=None)

desktop.release_desktop()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.11.7
asttokens==2.0.5
async-timeout==4.0.2
attrs==21.4.0
autopep8==1.6.0
Babel==2.10.3
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.6.15
cffi==1.15.1
charset-normalizer==2.1.0
clr-loader==0.1.7
colorama==0.4.5
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
deprecation==2.1.0
dill==0.3.5.1
entrypoints==0.4
executing==0.8.3
fastjsonschema==2.16.1
fonttools==4.34.4
frozenlist==1.3.0
glcontext==2.3.6
idna==3.3
imageio==2.19.5
importlib-metadata==4.12.0
ipykernel==6.15.1
ipython==8.4.0
ipython-genutils==0.2.0
ipywidgets==7.7.1
isort==5.10.1
jedi==0.18.1
Jinja2==3.1.2
json5==0.9.8
jsonschema==4.7.2
jupyter-client==7.3.4
jupyter-core==4.11.1
jupyter-packaging==0.12.2
jupyter-server==1.18.1
jupyterlab==3.4.4
jupyterlab-pygments==0.2.2
jupyterlab-server==2.15.0
jupyterlab-widgets==1.1.1
kiwisolver==1.4.4
lazy-object-proxy==1.7.1
lxml==4.9.1
MarkupSafe==2.1.1
matplotlib==3.5.2
matplotlib-inline==0.1.3
mccabe==0.7.0
mistune==0.8.4
moderngl==5.6.4
multidict==6.0.2
multipledispatch==0.6.0
nbclassic==0.4.3
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
notebook==6.4.12
notebook-shim==0.1.0
numpy==1.23.1
numpy-stl==2.17.1
open3d==0.15.1
packaging==21.3
pandas==1.4.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.2.0
platformdirs==2.5.2
plumbum==1.7.2
plyfile==0.7.4
prometheus-client==0.14.1
prompt-toolkit==3.0.30
psutil==5.9.1
pure-eval==0.2.2
pyaedt==0.5.9
pycodestyle==2.9.1
pycparser==2.21
Pygments==2.12.0
pylint==2.14.5
pyparsing==3.0.9
pyrr==0.10.3
pyrsistent==0.18.1
python-dateutil==2.8.2
python-pptx==0.6.21
python-utils==3.3.3
pythonnet==3.0.0rc4
pytorch3d==0.0.1
pytz==2022.1
pyvista==0.35.2
pywin32==304
pywinpty==2.0.6
pyzmq==23.2.0
requests==2.28.1
rpyc==5.0.1
scipy==1.8.1
scooby==0.5.12
seaborn==0.11.2
Send2Trash==1.8.0
simple-3dviz==0.5.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.2.post1
stack-data==0.3.0
terminado==0.15.0
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.1
tornado==6.2
traitlets==5.3.0
typing_extensions==4.3.0
urllib3==1.26.10
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.3
widgetsnbextension==3.6.1
wrapt==1.14.1
wslink==1.6.6
wxPython==4.2.0
XlsxWriter==3.0.3
yarl==1.7.2
zipp==3.8.1

maxcapodi78, 2022-08-23:
@wdmaiansys can you try the following?

```
hfss.export_3d_model('model','E:\\',fileFormat='.step', object_list=None,removed_objects=None)
```

I agree with you that stl should be handled and exception raised.

wdmaiansys, 2022-08-23:
Hi,

Thanks for the update.
But it still returns an error message:

Method export_3d_model Failed: General or AEDT Error. Please Check again (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024382), None) Project: 77174_V7_SPB16_parasolid, Design: Conformal (Terminal Network), [error] Script macro error: Command execution error, please see details in message window. (10:53:57 AM Aug 23, 2022) Arguments Provided: fileName = model filePath = E:\ fileFormat = .step object_list = None removed_objects = None Error in : File "c:\Python39\lib\site-packages\pyaedt\application\Analysis3D.py", line 530, in export_3d_model Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=export_3d_model
pyaedt error: Method export_3d_model Failed: General or AEDT Error. Please Check again pyaedt error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024382), None) pyaedt error: Project: 77174_V7_SPB16_parasolid, Design: Conformal (Terminal Network), [error] Script macro error: Command execution error, please see details in message window. (10:53:57 AM Aug 23, 2022) pyaedt error: Arguments Provided: pyaedt error: fileName = model pyaedt error: filePath = E:\ pyaedt error: fileFormat = .step pyaedt error: object_list = None pyaedt error: removed_objects = None pyaedt error: Error in : File "c:\Python39\lib\site-packages\pyaedt\application\Analysis3D.py", line 530, in export_3d_model pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=export_3d_model
False

Regards,

William

[External Sender]

maxcapodi78, 2022-08-26:
@wdmaiansys Which AEDT Version are you using? Is the export successfully working when you manually export from AEDT User Interface to step file?  I tried with a couple of models and worked for me

Samuelopez-ansys, 2022-11-11:
@wdmaiansys is the error still happening? Please contact us directly if it is. I will close this issue because we were not able to reproduce it.

closed 2022-11-11 by Samuelopez-ansys