ansys/pyaedt #1555: When running analyze_setup(), when no arguments are given, run the active setup

obrienpja, 2022-08-17:
### Description of the feature

Could you please change the signature of the analyze_setup() method so that setupname is not required, and that if setupname is not given, just run the active setup. Thanks!

priti-ashvin-shah-ibm, 2022-08-19:
Tagging so I can see the responses.

ring630, 2022-08-19:
@obrienpja I suppose you are referring to a hfss design. there is a method analysis_all() which runs all the enabled setups.

```
hfss = Hfss(specified_version="2022.2", non_graphical=non_graphical, designname="A1")
...
hfss.analyze_all()
```

obrienpja, 2022-08-24:
Hi @ring630, I am aware of this method, but I guess what you are saying is that there isn't really an "active" setup in the same way that there is an active design, but instead there are only enabled and disabled setups. Is that correct? I can close this issue if that is the case.

maxcapodi78, 2022-08-26:
@obrienpja there is already a method called analyze_nominal. this solves the first setup in the list.

ring630, 2022-08-26:
@obrienpja exactly, no "active setup". analyse_nominal is another option as @maxcapodi78 mentioned.

closed 2022-08-29 by MaxJPRey