ansys/pyaedt #1556: Get capacitance matrix from PostProcessor after Q3D simulation

obrienpja, 2022-08-17:
### Description of the feature

We are unable to get the capacitance matrix from the post processor or the "matrices" member of the Q3D object after running a Q3D simulation. Please provide a way for us to access this data in memory. Thanks!

priti-ashvin-shah-ibm, 2022-08-19:
I am tagging so I can see the response.

obrienpja, 2022-08-25:
To be clear, I am trying to understand if there is a way to get the capacitance matrix in memory without first exporting to file and then importing it.

maxcapodi78, 2022-08-31:
Hi @obrienpja yes it is.
 please have a look at this example

https://aedt.docs.pyansys.com/examples/05-Q3D/Q3D_Example.html#sphx-glr-examples-05-q3d-q3d-example-py

the get_solution_data does what you need.

closed 2022-08-31 by maxcapodi78

ThomasGM4, 2022-08-31:
Hi @maxcapodi78 
We were looking through that example but weren't able to sus out a means to extract the actual numerical values of the capacitance matrix. 

That example is what we had been working through, but couldn't see the means to get the matrix from q.matrices[0]

Were we just being blind to an obvious variable? We know we could export to txt/csv and then read that in.

priti-ashvin-shah-ibm, 2022-08-31:
Please open the issue since our questions for information has not been resolved.