ansys/pyaedt #1557: how can we change port_name

kclme84, 2022-08-18:
### Description of the feature

if we use h3d.create_edge_port("microstrip",0) to create one edge port ,how can we change this port name? thanks !

ring630, 2022-08-18:
@kclme84 changing property in aedt is not impossible, but cumbersome. I highly recommend to assign the correct port name in the first place.

ring630, 2022-08-18:
@kclme84, aedt is only intended to create. I am investigating if port can be created in EDB in which changing its properties is much more convenient.

kclme84, 2022-08-19:
@ring630 ok,thanks ! I will be trying!

kclme84, 2022-08-19:
we can use "Modeler3DLayout.change_property()" to modify the port name.thanks!
@ring630

closed 2022-08-19 by kclme84