ansys/pyaedt #1562: Add the option of "Convex Hull" in the argument of extent_type in the method create_cutout()

SHF101202021, 2022-08-19:
### Description of the feature

In HFSS 3D Layout, one expansion type of the cutout regions is "convex hull"
[image]

I want to request if this option can be added in ```Edb.create_coutout()``` or not
Though in ```Ansys.Ansoft.EDB.Geometry```, only ```BoundingBox``` and ```Conforming``` are available...
[image]

ring630, 2022-08-19:
@SHF101202021 If Convex Hull is not in Edb.Geometry, we cannot have it in PyAEDT.

ring630, 2022-08-19:
@svandenb-dev , is it possible any enhancement can be done in EDB.dll?

ring630, 2022-08-19:
@SHF101202021 I consider it as a limitation because it cannot be improved from pyaedt side. If it is ok, I will close this issue.

SHF101202021, 2022-08-19:
Ok. Please close this issue
Thank for your replies

ring630, 2022-08-23:
@SHF101202021 very interesting. I did some test but failed. HFSSExtentInfoType is not accepted by GetExpandedExtentFromNets. HFSSExtentInfoType is for other purposes. ConvexHull support as ExtentType needs to be implemented in Edb.dll. Let me discuss it internally.

SHF101202021, 2022-08-23:
@ring630 
Just share my idea:

Noted: ```aedb``` -> [image]

Before actually cutting out, one needs to define the polygon for the regions
[image]

Then, turn this polgon to be ```IList<PolyData>``` and implement ```Ansys.Ansoft.Geometry.PolygonData.GetConvexHullOfPolygons()``` like the following
[image]

It can pass eventually, however the arc data is missing: 
#### Polygon generated by ```GetExpandedExtentFromNets```
##### Code 
[image]
##### Shape
[image]

#### Polgygon transformed by ```GetConvexHullOfPolygons```
##### Code 
[image]
##### Shape
[image]

> If just observing the new shape, it is really convex hull, but without arc line if one save the data and see in 3DLayout.

### In 3D Layout

Here are the cut layout showed in 3D Layout

#### Cutout with ```GetConvexHullOfPolygons```
[image]

#### Cutout with 3DLayout Cutout GUI
[image]

SHF101202021, 2022-08-23:
> @SHF101202021 very interesting. I did some test but failed. HFSSExtentInfoType is not accepted by GetExpandedExtentFromNets. HFSSExtentInfoType is for other purposes. ConvexHull support as ExtentType needs to be implemented in Edb.dll. Let me discuss it internally.

@ring630  
Ansys had ever publish a tool called 'PinToPin.exe'
Maybe you can refer codes in Ansys internal Libraray
 as the guidance for this issue [image]

SHF101202021, 2022-08-24:
@ring630  
Got your point. 
Wait for your news for ConvexHull :D

ring630, 2022-08-25:
@SHF101202021 convex hull is implemented as you suggested. The generated extent is not exactly the same as it is from GUI, but I think it is acceptable.
[image]

SHF101202021, 2022-08-25:
@ring630 
Thanks for your trying!

closed 2022-08-25 by ring630