ansys/pyaedt #1564: Add one attributes "roughness" in "pyaedt.edb_core.EDB_Data.EDBLayer"

SHF101202021, 2022-08-19:
### Description of the feature

Now, only ```EDBLayer.roguhness_enabled```  is avilable in ```Edb```. However, one still needs to assign the roughness model for the in the stackup editor.
So, the ability to assign roughness models just in ```pyaedt.EDB``` is important for us

### Steps for implementing the feature

In ```Ansys.Ansoft.Edb``` data structure, roughness model is needed to be defined externally (See ```Ansys.Ansoft.Edb.Cell.RoughnessModel```).  

So, here is my poor idea: just add a method ```create_roughness_model``` in EDBLayer so that anyone can just assign the models in desired side (top, side, bottom)

```python
class EDBLayer(object):

    def __init__(self, edblayer, app):
        # Skip other definition
        
        ''''''
        self._top_roughness_model= None
        self._side_roughness_model = None
        self._bottom_roughness_model = None
        ''''''

        
        
    @pyaedt_function_handler()
    def create_roughness_model(
        self, 
        name,
        model_type="hurray",
        nodule_radius="10um", 
        surface_ratio="2.9"
        roughness=None,
        ):
        
        if model_type == "hurray":
            _roughness_model = self._edb.Cell.HurrayRoughnessModel(
                    noduleRadius=aedb.Utility.Value(nodule_radius),
                    surfaceRatio=aedb.Utility.Value(surface_ratio)
                )
        
        elif model_type == "groisse":
            _roughness_model = self._edb.Cell.GroisseRoughnessModel (
                    roguhness=aedb.Utility.Value(roughness),
                )
        
        if _roughness_model:
            return True
        else:
            return False        
        
        
    @property
    def top_roughness_model(self):
        model = self._top_roughness_model
        if "HurrayRoughnessModel" in model.ToString():

            return dict(model_type="hurray roghness model",
                        nodule_radius=model.NoduleRadius.ToString(),
                        surface_ratio=model.SurfaceRatio.ToString())
        elif "GroissRoughnessModel" in model.ToString():
            return dict(model_type="hurray roghness model",
                        roughness=model.Roughness.ToString(),
                        )
    
    @top_roughness_model.setter
    def top_roughness_model(name,
            model_type="hurray",
            nodule_radius="10um", 
            surface_ratio="2.9"
            roughness=None,
        ):
        _model = create_roughness_model(name, model_type, nodule_radius, surface_ratio, roughness)
        self._top_roughness_model = _model
        self.update_layers()
    
# =============================================================================
#     Side and Bottom do the same stuff
# =============================================================================
   @pyaedt_function_handler()
    def update_layer_vals(
        self,
        layerName,
        newLayer,
        etchMap,
        materialMap,
        fillMaterialMap,
        thicknessMap,
        negativeMap,
        roughnessMap,
        layerTypeMap,
    ):
        """"""
        # Skip
        if roughnessMap:
            newLayer.SetRoughnessEnabled(True)
            models = [self._top_roughness_model,
                     self._side_roughness_model,
                     self._bottom_roughness_model]            
            regions = [self._edb.Cell.RoughnessModel.Region.Top,
                       self._edb.Cell.RoughnessModel.Region.Side,
                       self._edb.Cell.RoughnessModel.Region.Bottom]
            
            # Setting the roughness in each sides
            for mdl, rgn in zip(models, regions):
                if nam:
                    newLayer.SetRoughnessModel(rgn, mdl)
        else:
            newLayer.SetRoughnessEnabled(False)
            
        # Skip

```

ring630, 2022-08-19:
@SHF101202021 you already have a good idea here. Will you able to create a branch and implemented?

maxcapodi78, 2022-08-23:
good idea. Thanks @ring630 for taking it

closed 2022-08-24 by maxcapodi78