ansys/pyaedt #1573: Bug located in pyaedt.modeler.Object3d.Padstack.layers

SHF101202021, 2022-08-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When using ```Hfss3dLayout(specified_version="2022.2")``` application, after loading the aedb folder, the ```hfss.modeler.padstacks[<padstack name>].layers``` shows empty dictionary ( See In [13] that all of the layers are empty dictionary)
[image]

### Steps To Reproduce

You may just load arbitrary aedb. 

### My thoughs for this bug

One needs to modify two parts in pyaedt

(1) ```pyaedt.modeler.Primitives3DLayout.Padstacks```:   
When pyaedt load aedb, for padstacks parts, program will get all padstack information by ```self.opadstackmanager.GetData```. Then, analyzing the data structure. However, when analyzing data starting with ```NAME:pds```, it seems for pyaedt to assume existing layers so the program doesn't ```add_layer``` when the index of the layers are one (see red part in the picture)
[image]

(2) ```pyaedt.modeler.Object3d.Padstack.add_layer```
In (1),  ```lay_id``` is not used.  In ```add_layer()```, the ```id``` is self-defined in the ````pyaedt.modeler.Object3D.Padstack``` rather than assigning the the same index of the original data
[image]

### My solutions for this bug

I modified two parts
(1) Add an extra argument  ```layer_id``` in ```add_layer``` and assign to the new ```pyaedt.modeler.Object3d.PDSlayer```
[image]

(2) Cancel the if condition in ```pyaedt.modeler.Primitives3DLayout.Padstacks```
[image]

After these two modifications, I can get layers in padstack
[image]

### Modified Codes
[3DLayout.zip](https://github.com/pyansys/pyaedt/files/9389236/3DLayout.zip)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
anaconda-client @ file:///C:/ci/anaconda-client_1635342725944/work
anaconda-navigator==2.1.1
anaconda-project @ file:///tmp/build/80754af9/anaconda-project_1626085644852/work
anyio @ file:///C:/ci/anyio_1620153135622/work/dist
appdirs==1.4.4
argh==0.26.2
argon2-cffi @ file:///C:/ci/argon2-cffi_1613037869401/work
arrow @ file:///C:/ci/arrow_1617738834352/work
asn1crypto @ file:///tmp/build/80754af9/asn1crypto_1596577642040/work
astroid @ file:///C:/ci/astroid_1628063282661/work
astropy @ file:///C:/ci/astropy_1629829318700/work
async-generator @ file:///home/ktietz/src/ci/async_generator_1611927993394/work
async-timeout==4.0.2
atomicwrites==1.4.0
attrs @ file:///tmp/build/80754af9/attrs_1620827162558/work
autopep8 @ file:///tmp/build/80754af9/autopep8_1620866417880/work
[... 255 lines cut ...]
urllib3==1.26.7
vtk==9.1.0
watchdog @ file:///C:/ci/watchdog_1624955113064/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websockets @ file:///C:/Users/user1/Desktop/PythonLibrary/websockets-10.2.tar.gz
Werkzeug @ file:///tmp/build/80754af9/werkzeug_1635505089296/work
whichcraft @ file:///tmp/build/80754af9/whichcraft_1617751293875/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1607531582688/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/ci/wrapt_1607574570428/work
wslink==1.6.6
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///tmp/build/80754af9/xlsxwriter_1628603415431/work
xlwings==0.24.9
xlwt==1.3.0
xmltodict @ file:///Users/ktietz/demo/mc3/conda-bld/xmltodict_1629301980723/work
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl==1.8.1
zict==2.0.0
zipp @ file:///tmp/build/80754af9/zipp_1633618647012/work
zope.event==4.5.0
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

maxcapodi78, 2022-08-23:
@SHF101202021 I saw you already proposed a solution. Would you be confortable in creating a PR?

maxcapodi78, 2022-08-30:
@SHF101202021 if you contact me on Teams I can teach you. It's pretty easy

closed 2022-08-30 by maxcapodi78

SHF101202021, 2022-09-03:
@maxcapodi78 
Great. 
Could you give me any contacting information?