ansys/pyaedt #1577: Bug located in Edb(edbpath=edb_path, edbversion="2022.1")

linmingchih, 2022-08-22:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The command works in 2022.1, but fails in 2022.2

### Steps To Reproduce

from pyaedt import Edb, Hfss3dLayout

edb_path = "D:/demo2/Galileo_G87173_204.brd"
edb = Edb(edbpath=edb_path, edbversion="2022.2")
print(edb.core_nets.nets.keys())

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
alabaster==0.7.12
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.2
astroid==2.11.7
async-timeout==4.0.2
atomicwrites==1.4.1
attrs==22.1.0
autopep8==1.6.0
Babel==2.10.3
backcall==0.2.0
bcrypt==3.2.2
beautifulsoup4==4.11.1
binaryornot==0.4.4
black==22.6.0
bleach==5.0.1
certifi==2022.6.15
cffi==1.15.1
chardet==5.0.0
charset-normalizer==2.1.0
click==8.1.3
cloudpickle==2.1.0
clr-loader==0.1.7
colorama==0.4.5
cookiecutter==2.1.1
cryptography==37.0.4
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20200713
dill==0.3.5.1
docutils==0.19
entrypoints==0.4
fastjsonschema==2.16.1
flake8==4.0.1
fonttools==4.34.4
frozenlist==1.3.0
idna==3.3
imageio==2.21.0
imagesize==1.4.1
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.15.1
ipython==7.34.0
ipython-genutils==0.2.0
isort==5.10.1
jedi==0.18.1
jellyfish==0.9.0
Jinja2==3.1.2
jinja2-time==0.2.0
json5==0.9.9
jsonschema==4.9.0
jupyter-client==7.3.4
jupyter-core==4.11.1
jupyter-server==1.18.1
jupyterlab==3.4.4
jupyterlab-pygments==0.2.2
jupyterlab-server==2.15.0
keyring==23.7.0
kiwisolver==1.4.4
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.5.2
matplotlib-inline==0.1.3
mccabe==0.6.1
mistune==0.8.4
multidict==6.0.2
mypy-extensions==0.4.3
nbclassic==0.4.3
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
notebook==6.4.12
notebook-shim==0.1.0
numpy==1.23.1
numpydoc==1.4.0
packaging==21.3
pandocfilters==1.5.0
paramiko==2.11.0
parso==0.8.3
pathspec==0.9.0
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.2.0
platformdirs==2.5.2
pluggy==1.0.0
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.30
psutil==5.9.1
ptyprocess==0.7.0
pyaedt==0.5.20
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.12.0
pylint==2.14.5
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.7
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.0
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
pyrsistent==0.18.1
python-dateutil==2.8.2
python-lsp-black==1.2.1
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.5.0
python-slugify==6.1.2
pythonnet==3.0.0rc4
pytoolconfig==1.2.2
pytz==2022.1
pyvista==0.36.1
pywin32==304
pywin32-ctypes==0.2.0
pywinpty==2.0.6
PyYAML==6.0
pyzmq==23.2.0
QDarkStyle==3.0.3
qstylizer==0.2.1
QtAwesome==1.1.1
qtconsole==5.3.1
QtPy==2.1.0
requests==2.28.1
rope==1.3.0
rpyc==5.0.1
Rtree==1.0.0
scooby==0.6.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.2.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.3.2.post1
Sphinx==5.1.1
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder==5.3.2
spyder-kernels==2.3.2
terminado==0.15.0
text-unidecode==1.3
textdistance==4.3.0
three-merge==0.1.1
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.1
tornado==6.2
traitlets==5.3.0
ujson==5.4.0
urllib3==1.26.11
vtk==9.2.0rc2
watchdog==2.1.9
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.3
whatthepatch==1.0.2
wrapt==1.14.1
wslink==1.6.6
yapf==0.32.0
yarl==1.8.0

SHF101202021, 2022-08-22:
Interesting! Look like python envirnment might cause the bug(?
My python is 3.9.7 and I can run your command without the bug
[image]

maxcapodi78, 2022-08-23:
@linmingchih if you recently installed aedt 2022.2 maybe is good to reboot the machine. I noticed in the past that sometime the python enviroment doesn't catch the newly created env variables in realtime

closed 2022-08-26 by maxcapodi78