ansys/pyaedt #1580: Fail to initialize Hfss when launched through SSH session

fcogalarza, 2022-08-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I'm trying to run an script that will build and simulate an HFSS  project. The script with the model has been tested and works jut fine. However, I'll like to automate a process to send multiple of this scripts to a local server where the models will be simulated and for that I'm  trying to use and SSH connection (transfer the script  with the model and launch on the server). The problem is I keep getting an error while launching Pyaedt, it shows the message below:

```
>>> hfss = Hfss("test", non_graphical=True)
pyaedt info: Logger Started
pyaedt info: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt info: AEDT installation Path C:\Program Files\AnsysEM\v222\Win64.
pyaedt info: Launching AEDT with module PythonNET.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\fjgalarz\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\hfss.py", line 164, in __init__
    FieldAnalysis3D.__init__(
  File "C:\Users\fjgalarz\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\application\Analysis3D.py", line 88, in __init__
    Analysis.__init__(
  File "C:\Users\fjgalarz\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\application\Analysis.py", line 98, in __init__
    Design.__init__(
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\application\Design.py", line 170, in __init__
    desktop = Desktop(
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\desktop.py", line 404, in __init__
    self._init_cpython(
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\site-packages\pyaedt\desktop.py", line 647, in _init_cpython
    self.logger.info("{} Started with process ID {}.".format(version, proc[0]))
IndexError: list index out of range
```
I get the same error message either running the script with the model or just by starting Hfss on a new python console. 

### Steps To Reproduce

Try to start a new Hfss model in a python console  on a remote host through ssh. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

`aiohttp==3.8.1
aiosignal==1.2.0
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asttokens==2.0.7
async-timeout==4.0.2
attrs==22.1.0
Babel==2.10.3
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.6.15
cffi==1.15.1
charset-normalizer==2.1.0
clr-loader==0.1.7
colorama==0.4.5
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
executing==0.9.1
fastjsonschema==2.16.1
fonttools==4.34.4
frozenlist==1.3.1
idna==3.3
imageio==2.21.1
importlib-metadata==4.12.0
ipykernel==6.15.1
ipython==8.4.0
ipython-genutils==0.2.0
jedi==0.18.1
Jinja2==3.1.2
json5==0.9.9
jsonschema==4.9.1
jupyter-client==7.3.4
jupyter-core==4.11.1
jupyter-server==1.18.1
jupyterlab==3.4.5
jupyterlab-pygments==0.2.2
jupyterlab-server==2.15.0
kiwisolver==1.4.4
lxml==4.9.1
MarkupSafe==2.1.1
matplotlib==3.5.2
matplotlib-inline==0.1.3
mistune==0.8.4
multidict==6.0.2
nbclassic==0.4.3
nbclient==0.6.6
nbconvert==6.5.2
nbformat==5.4.0
nest-asyncio==1.5.5
notebook==6.4.12
notebook-shim==0.1.0
numpy==1.23.1
packaging==21.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.2.0
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.30
psutil==5.9.1
pure-eval==0.2.2
pyaedt==0.5.20
pycparser==2.21
Pygments==2.12.0
pyparsing==3.0.9
pyrsistent==0.18.1
python-dateutil==2.8.2
pythonnet==3.0.0rc4
pytz==2022.1
pyvista==0.36.1
pywin32==304
pywinpty==2.0.7
pyzmq==23.2.0
requests==2.28.1
rpyc==5.0.1
scooby==0.6.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.2.post1
stack-data==0.3.0
terminado==0.15.0
tinycss2==1.1.1
tornado==6.2
traitlets==5.3.0
urllib3==1.26.11
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.3
wslink==1.6.6
yarl==1.8.1
zipp==3.8.1`

SHF101202021, 2022-08-23:
@fcogalarza 
What is your version of AEDT?
For example, my installtion version of AEDT is 2022 R2
Then, I need to use ```Hfss(specified_version="2022.2)``` to call 2022 R2 version of AEDT

I recommend to try again with specifying the installation version.

maxcapodi78, 2022-08-23:
maybe you should try to launch pyaedt using grpc api.
I've never tried it over ssh but it works well in a normal network.
on the remote machine you need a service running

`
from pyaedt.common_rpc import pyaedt_service_manager
pyaedt_service_manager()
`

On the client:

`
from pyaedt.common_rpc import create_session
cl = create_session("remotemachinename.domain.com")

from pyaedt import settings

settings.use_grpc_api = True

cl.aedt(port=50000, non_graphical=False)
hfss = Hfss(machine="remotemachinename.domain.com", port=50000)
# your code
`

fcogalarza, 2022-08-23:
Hi, thanks. I tested this approach and it fails on me due to not being able to write to Ansoft folder, so I guess it is a completely different issue. However, the functionality of this approach is a little different to what we need. This approach opens an AEDT instance on the server (also it opens in a graphical mode), so I guess sending multiple jobs simultaneously will cause issues, correct?

maxcapodi78, 2022-08-26:
@fcogalarza this approach opens a new AEDT session on the remote machine. The remote session can be graphical or non graphical and multiple instances on multiple ports can be run together.
Can you detail more what is the architecture you have in mind? Setup everything on the client and just submit to the server or do it on the server?

fcogalarza, 2022-08-29:
The main goal is to run a group of models with different structures, based on a given algorithm. For that, I'll be building different scripts with basically a model described using pyaedt. Since this is going to be an iterative process I was hoping to upload the script with the model and send it to run on the server. The "easiest" approach was to upload everything through SSH/SCP and just execute the script on the remote server, but that is the part that is failing. However, seems that grpc may be configured to do something similar if it can run non graphical and with multiple connections simultaneously. I'll study some more on that, thanks so much for your inputs.

MaxJPRey, 2022-08-30:
Hi @fcogalarza,
Yes, as described by @maxcapodi78 our gRPC implementation in PyAEDT should allow you to run your models and scripts on a remote server and handle several instances.
Let us know if you still encounter any issues.

gmalinve, 2023-02-16:
Hi @fcogalarza ,

Hopefully our suggestions helped you to overcome your issues.
Could you please let us know if we can close this issue?

Thanks!

Giulia

gmalinve, 2023-02-17:
Hi @fcogalarza ,

Please find here an explanation on how to launch PyAEDT on a remote machine which works for v2022.2 and higher:
https://aedt.docs.pyansys.com/release/0.6/Getting_started/ClientServer.html#grpc-connection-in-aedt-2022-r2-and-later

Hopefully this helps!
I'm going to close this issue since it has been open for a while.
In case you need further help please reopen a new one.

Thanks!

closed 2023-02-17 by gmalinve