ansys/pyaedt #1623: Add  an optional additional agrugment to assign_lumped_rlc_to_sheet.

priti-ashvin-shah-ibm, 2022-08-29:
### Description of the feature

In method , assign_lumped_rlc_to_sheet, allow the user to enter a start and stop point, then the user doesn't have to give you the axis.  

In another words, The use can provide the start and stop for props["CurrentLine"], instead of passing axisdir.

`props["CurrentLine"] = OrderedDict({"Start": start, "End": stop})`

https://aedt.docs.pyansys.com/API/_autosummary/pyaedt.Hfss.assign_lumped_rlc_to_sheet.html

### Steps for implementing the feature

Allow user a choice of passing either asixdir or start/stop 3-d points.

Samuelopez-ansys, 2022-09-01:
Hi @priti-ashvin-shah-ibm and @obrienpja ,

Almost all hfss methods with integration lines use the axisdir because it is safer. PyAEDT ensures the integration line defined is going to be valid because the Start and the End point are calculated using this method:

_point0, point1 = self.modeler.get_mid_points_on_dir(sheet_name, axisdir)_

Then I prefer to don't change this method, and if the user needs to change the default Start/End points you could modify it:

**_imp = aedtapp.assign_lumped_rlc_to_sheet(rect.name, 0s, Rvalue=50, Lvalue=1e-9)
imp.props["CurrentLine"]["Start"] = ["10.0mm", "2.0mm", "0.0mm"]_**

This will change the original point:
[image]

[image]

What do you think?

Regards

priti-ashvin-shah-ibm, 2022-09-01:
I see your point. I am sharing our point of view.  We already have the start and end points of all integration lines within HFSS.  Then we have to determine the axis to pass to pyaedt.  Then pyaedt calculates the start and end point again.  We were trying to be efficient; this is not a deal-breaker.

Is it possible to have error checking based on what is passed to pyaedt?

priti-ashvin-shah-ibm, 2022-09-01:
Also, all of our integration lines are not parallel to an axis.

priti-ashvin-shah-ibm, 2022-09-01:
Now, this could be a deal-breaker since I can't control the integration line as desired.

Samuelopez-ansys, 2022-09-05:
Hi @priti-ashvin-shah-ibm,

Which start_vec_xyz and end_vec_xyz are you testing?

I have tested:

start_vec_xyz = [0, 0, 0]

end_vec_xyz = [10, 20, 30]

And it worked:
[image]

What am I missing?

priti-ashvin-shah-ibm, 2022-09-06:
@Samuelopez-ansys 
I see that you have a pull request, would that work for both  create_lumped_port_to_sheet and assign_lumped_rlc_to_sheet?

Samuelopez-ansys, 2022-09-06:
@priti-ashvin-shah-ibm 

Indeed , it should work in both methods.

could you test it in your model?

Hopefully it will overcome the limitation of using axisdir.

You just need to pass a list of coordinates:
[[xstart, ystart, zstart], [xend, yend, zend]]

regards

priti-ashvin-shah-ibm, 2022-09-06:
Using

endpoints_3d
[[-0.195, 0.425, 0.0675], [-0.195, 0.431, 0.0675]]

```
  lumped_port = self.current_app.create_lumped_port_to_sheet(
                sheet_name=poly_port.name,
                # Name is under excitation folder, so can reuse.
                portname=poly_port.name,
                axisdir=endpoints_3d,
                impedance=impedance)
```

Got error:

[image]

priti-ashvin-shah-ibm, 2022-09-06:
Using 
```
     poly_jj = self.current_app.modeler.create_polyline(
            [endpoints_3d[0], endpoints_3d[1]],
            name=poly_jj_name,
            close_surface=False)
        poly_jj.show_arrow = True
        poly_jj.color = (128, 0, 128)
```

endpoints_3d.tolist()
[[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]

    lumped_rlc_boundary = self.current_app.assign_lumped_rlc_to_sheet(
                sheet_name=poly_ansys.name,
                sourcename=f'rlc_{poly_ansys.name}',
                axisdir=endpoints_3d.tolist(),
                Lvalue=10e-9)

Got error:
[image]

priti-ashvin-shah-ibm, 2022-09-06:
@Samuelopez-ansys , I may have entered the syntax incorrectly for axisdir.  I will look at the documentation again.

Samuelopez-ansys, 2022-09-07:
Hi @priti-ashvin-shah-ibm ,

You need to pass a list of list. Axisdir admits 3 types, integer and pyaedt.AxisDir if you want to use the axis direction or a list in case you want to specify the coordinates explicitly, then you just need to pass the list of list : [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]].

I tried your code and it worked:

endpoints_3d = [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]

poly_ansys = hfss.modeler.create_polyline(
            [endpoints_3d[0], endpoints_3d[1]],
            name="name",
            close_surface=False,
            xsection_type='Line',
            xsection_width="2mm")

lumped_rlc_boundary = hfss.assign_lumped_rlc_to_sheet(
            sheet_name=poly_ansys.name,
            sourcename=f'rlc_{poly_ansys.name}',
            axisdir=endpoints_3d,
            Lvalue=10e-9)

[image]

priti-ashvin-shah-ibm, 2022-09-07:
I am glad that we are working together to get a working solution!!!.  There is something that is happening different between us.  I would like passing a list to axisdir to work with both drivenmodal and eigenmode solution types for HFSS.  Presently, I am using drivenmodal.   I am using Ansys 2021 R2

Hmmm, in both scenarios I shared above, I passed a list of list.  That is shown in both of the error messages that I shared above.  However, I will re-run the code and explicitly printed out the type of data that I passed to axisdir.  

Below are the error messages that I got for assign_lumped_rlc_to_sheet and create_lumped_port_to_sheet.

This is in both error messages.
**_pyaedt error: unsupported operand type(s) for divmod(): 'list' and 'int'_**

Below are the details of both scenarios.
***********************************Using: assign_lumped_rlc_to_sheet
```
poly_ansys = self.current_app.modeler.create_polyline(
            endpoints_3d,
            #[endpoints_3d[0], endpoints_3d[1]],
            name=jj_name,
            close_surface=False,
            xsection_type='Line',
            xsection_width=qc_width)

print(f'The type for endpioint_3d = {type(endpoints_3d)}')
print(f'The type for endpioint_3d[0] = {type(endpoints_3d[0])}')
print(f'The type for endpioint_3d[1] = {type(endpoints_3d[1])}')
lumped_rlc_boundary = self.current_app.assign_lumped_rlc_to_sheet(
                sheet_name=poly_ansys.name,
                sourcename=f'rlc_{poly_ansys.name}',
                axisdir=endpoints_3d,
                Lvalue=10e-9)
```

The output:
The type for endpioint_3d = <class 'list'>
The type for endpioint_3d[0] = <class 'list'>
The type for endpioint_3d[1] = <class 'list'>
pyaedt error: Method get_mid_points_on_dir Failed:  Type Error. Please Check again
Method get_mid_points_on_dir Failed:  Type Error. Please Check again
_**pyaedt error: unsupported operand type(s) for divmod(): 'list' and 'int'**_
unsupported operand type(s) for divmod(): 'list' and 'int'
pyaedt error: Arguments Provided:
Arguments Provided:
pyaedt error:     sheet = JJ_rect_Lj_4_poly4
    sheet = JJ_rect_Lj_4_poly4
**_pyaedt error:     **axisdir = [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]**_**
    axisdir = [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]
pyaedt error: Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\modeler\Primitives.py", line 2646, in get_mid_points_on_dir
Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\modeler\Primitives.py", line 2646, in get_mid_points_on_dir
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=get_mid_points_on_dir
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=get_mid_points_on_dir
pyaedt error: Method assign_lumped_rlc_to_sheet Failed:  Type Error. Please Check again
Method assign_lumped_rlc_to_sheet Failed:  Type Error. Please Check again
pyaedt error: cannot unpack non-iterable bool object
cannot unpack non-iterable bool object
pyaedt error: Arguments Provided:
Arguments Provided:
pyaedt error:     sheet_name = JJ_rect_Lj_4_poly4
    sheet_name = JJ_rect_Lj_4_poly4
pyaedt error:     axisdir = [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]
    axisdir = [[0.4, 1.015, 0.005], [0.4, 1.05, 0.005]]
pyaedt error:     sourcename = rlc_JJ_rect_Lj_4_poly4
    sourcename = rlc_JJ_rect_Lj_4_poly4
pyaedt error:     rlctype = None
    rlctype = None
pyaedt error:     Rvalue = None
    Rvalue = None
pyaedt error:     Lvalue = 1e-08
    Lvalue = 1e-08
pyaedt error:     Cvalue = None
    Cvalue = None
pyaedt error: Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\hfss.py", line 3535, in assign_lumped_rlc_to_sheet
Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\hfss.py", line 3535, in assign_lumped_rlc_to_sheet
[... 44 lines cut ...]
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=get_mid_points_on_dir
pyaedt error: Method create_lumped_port_to_sheet Failed:  Type Error. Please Check again
Method create_lumped_port_to_sheet Failed:  Type Error. Please Check again
pyaedt error: cannot unpack non-iterable bool object
cannot unpack non-iterable bool object
pyaedt error: Arguments Provided:
Arguments Provided:
pyaedt error:     sheet_name = Port_Q1_readout
    sheet_name = Port_Q1_readout
pyaedt error:     axisdir = [[-0.195, 0.425, 0.0675], [-0.195, 0.431, 0.0675]]
    axisdir = [[-0.195, 0.425, 0.0675], [-0.195, 0.431, 0.0675]]
pyaedt error:     impedance = 50
    impedance = 50
pyaedt error:     portname = Port_Q1_readout
    portname = Port_Q1_readout
pyaedt error:     renorm = None
    renorm = None
pyaedt error:     deemb = None
    deemb = None
pyaedt error:     reference_object_list = None
    reference_object_list = None
pyaedt error: Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\hfss.py", line 3251, in create_lumped_port_to_sheet
Error in :   File "C:\Users\priti.ashvin.shah\source\repos\pyaedt\pyaedt\hfss.py", line 3251, in create_lumped_port_to_sheet
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_lumped_port_to_sheet
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_lumped_port_to_sheet

Samuelopez-ansys, 2022-09-07:
Really weird @priti-ashvin-shah-ibm,

Did you checkout the branch? Or are you using the main branch? Because hte Pull request is still active and it is not merged to the main.

PR #1655 (2022-09-08): Define axisdir with points explictly
Fix #1623 

Sometimes is not clear how the integration line is defined, mainly, when the sheet is not parallel to the CS. I think these methods should have the option of defining explicitly the coordinates of the integration lines.