ansys/pyaedt #1646: Unable to install PyAEDT with Poetry 1.2.0

iazehaf, 2022-09-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to install pyaedt with **Poetry 1.2.0**. 

In my pyproject.toml I have only one dependency:
```
[tool.poetry.dependencies] 
python = ">=3.7,<3.9"
pyaedt = {version = "*"} 
```

I have no private source declared in the pyproject.toml nor in the poetry configuration. I expect Poetry to collect pyAEDT from the public PyPI.

Error received:
```
  PoetryException

  Failed to install C:/Users/iazehaf/AppData/Local/pypoetry/Cache/artifacts/a7/b8/f6/0d12ea2ceebcbc499f55e481bd52e306181f5601111ef50d8e08857bab/pyaedt-0.5.40.tar.gz

  at .venv\lib\site-packages\poetry\utils\pip.py:51 in pip_install
       47│
       48│     try:
       49│         return environment.run_pip(*args)
       50│     except EnvCommandError as e:
    →  51│         raise PoetryException(f"Failed to install {path.as_posix()}") from e
       52│
```

Note that if I try with Poetry 1.1.15 it works fine. 

### Steps To Reproduce

Commands executed:
- python -m venv .venv
- .venv\Scripts\python-m pip install --upgrade pip
- .venv\Scripts\python-m pip install --upgrade poetry
- .venv\Scripts\poetry install

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
anyio==3.6.1
appdirs==1.4.4
argcomplete==2.0.0
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0     
async-timeout==4.0.2
asynctest==0.13.0
atomicwrites==1.4.0
attrs==21.4.0
Babel==2.10.3
backcall==0.2.0
beautifulsoup4==4.11.1
black==22.3.0
bleach==5.0.1
Brotli==1.0.9
build==0.8.0
CacheControl==0.12.11
cachelib==0.9.0
cachy==0.3.0
cefpython3==66.1
certifi==2021.10.8
cffi==1.15.1
charset-normalizer==2.0.12       
chromedriver-autoinstaller==0.4.0
cleo==0.8.1
click==8.1.3
clikit==0.6.2
colorama==0.4.4
commonmark==0.9.1
coverage==6.4.1
crashtest==0.3.1
cycler==0.11.0
dash==2.5.0
dash-bootstrap-components==1.0.3
dash-core-components==2.0.0
dash-extensions==0.1.5
dash-html-components==2.0.0
dash-iconify==0.1.2
dash-table==5.0.0
debugpy==1.6.0
decorator==5.1.1
defusedxml==0.7.1
demo-package @ file:///D:/SandBox/02.Technical/01.Projects/03.GitHub_Migration/02.Technical/demo-package/dist/demo_package-0.1.0-py3-none-any.whl
distlib==0.3.4
docutils==0.18.1
EditorConfig==0.12.3
entrypoints==0.4
fastapi==0.75.0
fastjsonschema==2.15.3
filelock==3.7.1
Flask==2.1.2
Flask-Caching==2.0.0
Flask-Compress==1.12
fonttools==4.33.3
frozenlist==1.3.0
geckodriver-autoinstaller==0.1.0
graphviz==0.20
grpcio==1.42.0
grpcio-tools==1.42.0
h11==0.13.0
html5lib==1.1
idna==3.3
imageio==2.19.0
importlib-metadata==4.11.3
importlib-resources==5.8.0
iniconfig==1.1.1
ipykernel==6.15.0
ipython==7.33.0
ipython-genutils==0.2.0
isort==5.10.1
itsdangerous==2.1.2
jedi==0.18.1
Jinja2==3.1.2
jsbeautifier==1.14.5
json5==0.9.8
jsonschema==4.6.1
jupyter-client==7.3.4
jupyter-core==4.10.0
jupyter-server==1.18.0
jupyterlab==3.4.3
jupyterlab-pygments==0.2.2
jupyterlab-server==2.14.0
keyring==23.8.2
kiwisolver==1.4.2
lockfile==0.12.2
lxml==4.9.1
MarkupSafe==2.1.1
matplotlib==3.5.1
matplotlib-inline==0.1.3
meshio==4.4.6
mistune==0.8.4
more-itertools==8.14.0
msgpack==1.0.4
multidict==6.0.2
mypy-extensions==0.4.3
nbclassic==0.4.0
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
networkx==2.6.3
notebook-shim==0.1.0
numpy==1.21.6
packaging==21.3
pandas==1.3.5
pandocfilters==1.5.0
parso==0.8.3
pastel==0.2.1
pathspec==0.9.0
pep517==0.12.0
percy==2.0.2
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.1.0
pipenv==2022.6.7
pipx==1.1.0
pkginfo==1.8.3
platformdirs==2.5.2
plotly==5.6.0
pluggy==1.0.0
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.29
protobuf==3.19.1
proxy-tools==0.1.0
psutil==5.9.1
ptyprocess==0.7.0
py==1.11.0
pycparser==2.21
pydantic==1.9.1
pydepgraph==0.1
Pygments==2.12.0
pylev==1.4.0
pyparsing==3.0.8
pyrsistent==0.18.1
pyshortcuts==1.8.2
pytest==6.2.5
pytest-base-url==1.4.2
pytest-cov==3.0.0
pytest-html==3.1.1
pytest-metadata==2.0.2
pytest-selenium==3.0.0
pytest-variables==1.9.0
python-dateutil==2.8.2
python-multipart==0.0.5
pythonnet==2.5.2
pytz==2022.1
pyvista==0.34.1
pywebview==3.6.3
pywin32==304
pywin32-ctypes==0.2.0
pywinpty==2.0.5
pyzmq==23.2.0
readme-renderer==35.0
requests==2.27.1
requests-toolbelt==0.9.1
rfc3986==2.0.0
rich==12.4.4
rpyc==5.1.0
scipy==1.7.3
scooby==0.5.12
selenium==3.141.0
Send2Trash==1.8.0
shellingham==1.4.0
six==1.13.0
sniffio==1.2.0
soupsieve==2.3.2.post1
starlette==0.17.1
tenacity==6.3.1
terminado==0.15.0
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.0
tornado==6.2
tox==3.25.1
tox-gh-actions==2.9.1
traitlets==5.1.1
twine==4.0.1
typed-ast==1.5.4
typeguard==2.13.0
typing_extensions==4.2.0
urllib3==1.26.9
userpath==1.8.0
uvicorn==0.18.2
virtualenv==20.14.1
virtualenv-clone==0.5.7
vtk==9.1.0
waitress==2.1.2
wcwidth==0.2.5
webapppoc==0.0.1
webencodings==0.5.1
websocket-client==1.3.3
Werkzeug==2.1.2
wslink==1.6.1
yarl==1.7.2
zipp==3.8.0

iazehaf, 2022-10-04:
Hi,

I find this [thread](https://github.com/python-poetry/poetry/issues/6453) in GitHub which may be related to our issue.

In fact, Poetry prints a series of errors among which we can find this error:
`Command 'D:\SandBox\02.Technical\01.Temp\sf-saf\.venv\Scripts\python.exe -m pip install --use-pep517 --disable-pip-version-check --prefix D:\SandBox\02.Technical\01.Temp\sf-saf\.venv --no-deps C:\Users\iazehaf\AppData\Local\pypoetry\Cache\artifacts\b4\35\76\8c95d46ee440de7b3f2e5335f6e80029557807308f6e02474978b8012d\pyaedt-0.4.88.tar.gz' returned non-zero exit status 1.`

It seems like PyAEDT does not support an a PEP517 inistall.

nick-marnik, 2023-01-24:
For anyone else who faces this issue, there is a temporary workaround: install pyaedt as an editable package:
`poetry add --editable git+https://github.com/pyansys/pyaedt.git#v0.6.34`

Which gives you the following line in `pyproject.toml`:
`pyaedt = {git = "https://github.com/pyansys/pyaedt.git", rev = "v0.6.34", develop = true}`

This works because of the poetry implementation: https://github.com/python-poetry/poetry/blob/ef89e90fc1305e07a62e9e715c91a66c8d7425f7/src/poetry/utils/pip.py#L37-L38

tpgillam, 2023-02-28:
> For anyone else who faces this issue, there is a temporary workaround: install pyaedt as an editable package: `poetry add --editable git+https://github.com/pyansys/pyaedt.git#v0.6.34`
>
> Which gives [...]

As of poetry 1.4.0, it seems like this workaround no longer works, sadly. Here's the stack trace that I get when using it:

```python
bash> poetry add --editable git+https://github.com/pyansys/pyaedt.git#v0.6.48

Updating dependencies
Resolving dependencies... (0.2s)

Writing lock file

Package operations: 1 install, 0 updates, 0 removals

  • Installing pyaedt (0.6.48 90a54f7): Failed

  ChefBuildError

  Backend subprocess exited when trying to invoke get_requires_for_build_wheel

  Traceback (most recent call last):
    File "/home/tom/.local/share/pypoetry/venv/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
      main()
    File "/home/tom/.local/share/pypoetry/venv/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/tom/.local/share/pypoetry/venv/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/tmp/tmptgln3x_r/.venv/lib/python3.10/site-packages/setuptools/build_meta.py", line 338, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "/tmp/tmptgln3x_r/.venv/lib/python3.10/site-packages/setuptools/build_meta.py", line 320, in _get_build_requires
      self.run_setup()
    File "/tmp/tmptgln3x_r/.venv/lib/python3.10/site-packages/setuptools/build_meta.py", line 484, in run_setup
      super(_BuildMetaLegacyBackend,
    File "/tmp/tmptgln3x_r/.venv/lib/python3.10/site-packages/setuptools/build_meta.py", line 335, in run_setup
      exec(code, locals())
    File "<string>", line 5, in <module>
  ModuleNotFoundError: No module named 'pip'

  at ~/.local/share/pypoetry/venv/lib/python3.10/site-packages/poetry/installation/chef.py:152 in _prepare
      148│
      149│                 error = ChefBuildError("\n\n".join(message_parts))
      150│
      151│             if error is not None:
    → 152│                 raise error from None
      153│
      154│             return path
      155│
      156│     def _prepare_sdist(self, archive: Path, destination: Path | None = None) -> Path:
```

This is very similar to the stack trace we get when doing the naive `poetry add pyaedt` -- specifically both go via the `get_requires_for_build_wheel` function, which is unhappy.

nick-marnik, 2023-03-08:
I have to test locally, but I think when using poetry 1.4, one can use `poetry config installer.modern-installation false` to fix this error.

This is strictly a workaround - one of the biggest features of poetry 1.4 is the new modern installer that is supposed to greatly reduce the time required to resolve dependencies.

tpgillam, 2023-03-09:
Thanks @nick-marnik - for now we've just pinned to poetry 1.3.2, using the original workaround above. 

We're looking forward to #2539 , since I think this should fix this properly!

MaxJPRey, 2023-03-10:
Fixed in the PR #2539 .

closed 2023-03-10 by MaxJPRey