ansys/pyaedt #1650: We didn't get resolution from response in prior issue

priti-ashvin-shah-ibm, 2022-09-02:
### Description of the feature

The prior issue was closed without resolution of our question. 
 https://github.com/pyansys/pyaedt/issues/1556.

Could you please address the comment:
```
We were looking through that example but weren't able to sus out a means to extract the actual numerical values of the capacitance matrix.

That example is what we had been working through, but couldn't see the means to get the matrix from q.matrices[0]

Were we just being blind to an obvious variable? We know we could export to txt/csv and then read that in.
Or is it that get_solution_data(expressions=[data_plot_self]...) would need an expression that is for the matrix?
```

Since we are using an API that has the information in memory already,  It doesn't make sense to write to a file and read it.  It would be helpful to show in documentation how to clearly get out  all data using pyaedt.  

### Useful links and references

https://github.com/pyansys/pyaedt/issues/1556.

priti-ashvin-shah-ibm, 2022-09-02:
@obrienpja , @ThomasGM4

Samuelopez-ansys, 2022-09-02:
Hi @obrienpja @priti-ashvin-shah-ibm,

If I understood this issue, you need to access the C Matrix data right?

There are 2 methods to access this data without export the results to a CSV file.

The first method is the one we prefer in pyaedt:

from pyaedt import Q3d
app = Q3d(specified_version='2022.2')

all_C_matrices = app.matrices[0].get_sources_for_plot()

**# Get data
a = app.post.get_solution_data(expressions=all_C_matrices, context="Original")**

**# Method 1
a.active_expression # Default expression for example 'C(groundplane,groundplane)'
c1 = a.data_real()**

You just need to change the active expression to obtain the value:

**a.active_expression = all_C_matrices[1]
c2 = a.data_real()**

The second method you could use is access directly to the private method "_solution_data_real()" which it is used by the previous method ("data_real"):

**C_matrix_real = a._solution_data_real()**

The previous command returns a dictionary with the name of the different terms of the matrix and the value.

The units are saved here:

**c1_units = a.units_data**

Does it solve your question?

Regards

priti-ashvin-shah-ibm, 2022-09-02:
@Samuelopez-ansys  Thank you for your quick response!!! This is  helpful.  I will check  with other members of group, who will be available on Tuesday to confirm we have what we need.

ThomasGM4, 2022-09-02:
Brilliant, yep that looks like the expression we were looking for. Thanks @Samuelopez-ansys !

To explain the reasoning, we need the matrix to then pass specific values into the LOMAnalysis, eg. the capacitance between the two charge islands of the transmon.

priti-ashvin-shah-ibm, 2022-09-02:
@Samuelopez-ansys  Again, thanks for your help!!!! We can close this issue.

closed 2022-09-02 by priti-ashvin-shah-ibm