ansys/pyaedt #1654: Add method or property to get a list of designs in the currently open project from an instance of the Desktop() class.

dcrawforAtAnsys, 2022-09-05:
### Description of the feature

We can get a list of project names, but do not have access to the pyaedt designs and type of design (i.e. Hfss(), Maxwell(), ...)

### Steps for implementing the feature

Create a property that returns a list of application object instances.  This could be implemented through an iterator that behaves like a dictionary for the desktop object.  The keys would be project name and design name. For example:
desktop["Project1"]["hfss_design"] would return an instance of Hfss.

Samuelopez-ansys, 2022-09-26:
Hi @dcrawforAtAnsys ,

Are you working on that? If not, I will check how to implement it.

dcrawforAtAnsys, 2022-09-26:
Im not working on it. Please go ahead and take it

PR #1740 (2022-09-27): Return PyAEDT app object from Desktop object
Close #1654 and #1719
closed 2022-09-27 by maxcapodi78