ansys/pyaedt #1671: Add SIwave DCIR post processing features

skandak-ansys, 2022-09-07:
### Description of the feature

After performing DCIR simulation, there is no way for the user to extract information about Loop resistance, path resistance, RL table etc. It will be useful for the user to access these parameters through pyaedt commands.

Samuelopez-ansys, 2023-02-13:
@ring630 What is the status of this issue?

ring630, 2023-02-17:
@skandak-ansys new method added. 

`
hfss3d = Hfss3dLayout(self.dcir_example_project, "Galileo_G87173_204", specified_version=desktop_version)
hfss3d.get_dcir_solution_data("Siwave_DC_WP9QNY", "RL", "Path Resistance")
hfss3d.get_dcir_solution_data("Siwave_DC_WP9QNY", "Vias", "Current")

solution_data = hfss3d.get_dcir_solution_data("Siwave_DC_WP9QNY", "Sources", "Voltage")
solution_data.data_magnitude(solution_data.expressions[0])
`

svandenb-dev, 2023-02-20:
closing the issue thanks @ring630

closed 2023-02-20 by svandenb-dev