ansys/pyaedt #1677: Add ...Q3D The order of data that we would prefer is the same order that is provided, when one uses the option to print to csv. Not sure if this is available through pyaedt. Any insight into the order of data would be helpful.

priti-ashvin-shah-ibm, 2022-09-08:
### Description of the feature

The order of data that we would prefer in the capacitance matrix in a dict is the same order that is provided, when one uses the option to print to csv file. Not sure if this is available through pyaedt. Any insight into the order of data would be helpful. 

@obrienpja , @ThomasGM4 , @Samuelopez-ansys 

This is what we are presently doing to write solution data to file and read it back.  We would like to avoid writing and reading of data to a file, but be able to get solution data in a the similar format. 

From 
https://github.com/zlatko-minev/pyEPR/blob/5e5e0e161477206e69e06f3175c7c08d06a86110/pyEPR/ansys.py#L1528
```
def get_matrix(
            self,
            variation='',
            pass_number=0,
            frequency=None,
            MatrixType='Maxwell',
            solution_kind='LastAdaptive',  # AdaptivePass
            ACPlusDCResistance=False,
            soln_type="C"):
        '''
        Arguments:
        -----------
            variation: an empty string returns nominal variation.
                        Otherwise need the list
            frequency: in Hz
            soln_type = "C", "AC RL" and "DC RL"
            solution_kind = 'LastAdaptive' # AdaptivePass
        Internals:
        -----------
            Uses self.solution_name  = Setup1 : LastAdaptive

        Returns:
        ---------------------
            df_cmat, user_units, (df_cond, units_cond), design_variation
        '''
        if frequency is None:
            frequency = self.get_frequency_Hz()

        temp = tempfile.NamedTemporaryFile()
        temp.close()
        path = temp.name + '.txt'
        # <FileName>, <SolnType>, <DesignVariationKey>, <Solution>, <Matrix>, <ResUnit>,
        # <IndUnit>, <CapUnit>, <CondUnit>, <Frequency>, <MatrixType>, <PassNumber>,
        # <ACPlusDCResistance>
        logger.info(f'Exporting matrix data to ({path}, {soln_type}, {variation}, '
                                             f'{self.name}:{solution_kind}, '
                                             '"Original", "ohm", "nH", "fF", '
                                             f'"mSie", {frequency}, {MatrixType}, '
                                             f'{pass_number}, {ACPlusDCResistance}')
        self.parent._design.ExportMatrixData(path, soln_type, variation,
                                             f'{self.name}:{solution_kind}',
                                             "Original", "ohm", "nH", "fF",
                                             "mSie", frequency, MatrixType,
                                             pass_number, ACPlusDCResistance)

        df_cmat, user_units, (df_cond, units_cond), design_variation = \
            self.load_q3d_matrix(path)
        return df_cmat, user_units, (df_cond, units_cond), design_variation
```

```
  def _readin_Q3D_matrix(path: str):
        """
        Read in the txt file created from q3d export
        and output the capacitance matrix

        When exporting pick "save as type: data table"

        See Zlatko

        RETURNS: Dataframe

        Example file:
        ```
        DesignVariation:$BBoxL='650um' $boxH='750um' $boxL='2mm' $QubitGap='30um' \
                        $QubitH='90um' \$QubitL='450um' Lj_1='13nH'
        Setup1:LastAdaptive
        Problem Type:C
        C Units:farad, G Units:mSie
        Reduce Matrix:Original
        Frequency: 5.5E+09 Hz

        Capacitance Matrix
            ground_plane	Q1_bus_Q0_connector_pad	Q1_bus_Q2_connector_pad	Q1_pad_bot	Q1_pad_top1	Q1_readout_connector_pad
        ground_plane	2.8829E-13	-3.254E-14	-3.1978E-14	-4.0063E-14	-4.3842E-14	-3.0053E-14
        Q1_bus_Q0_connector_pad	-3.254E-14	4.7257E-14	-2.2765E-16	-1.269E-14	-1.3351E-15	-1.451E-16
        Q1_bus_Q2_connector_pad	-3.1978E-14	-2.2765E-16	4.5327E-14	-1.218E-15	-1.1552E-14	-5.0414E-17
        Q1_pad_bot	-4.0063E-14	-1.269E-14	-1.218E-15	9.5831E-14	-3.2415E-14	-8.3665E-15
        Q1_pad_top1	-4.3842E-14	-1.3351E-15	-1.1552E-14	-3.2415E-14	9.132E-14	-1.0199E-15
        Q1_readout_connector_pad	-3.0053E-14	-1.451E-16	-5.0414E-17	-8.3665E-15	-1.0199E-15	3.9884E-14

        Conductance Matrix
            ground_plane	Q1_bus_Q0_connector_pad	Q1_bus_Q2_connector_pad	Q1_pad_bot	Q1_pad_top1	Q1_readout_connector_pad
        ground_plane	0	0	0	0	0	0
        Q1_bus_Q0_connector_pad	0	0	0	0	0	0
        Q1_bus_Q2_connector_pad	0	0	0	0	0	0
        Q1_pad_bot	0	0	0	0	0	0
        Q1_pad_top1	0	0	0	0	0	0
        Q1_readout_connector_pad	0	0	0	0	0	0
        ```
        """

        text = Path(path).read_text()

        s1 = text.split('Capacitance Matrix')
        assert len(s1) == 2, "Could not split text to `Capacitance Matrix`"

        s2 = s1[1].split('Conductance Matrix')

        df_cmat = pd.read_csv(io.StringIO(s2[0].strip()),
                              delim_whitespace=True,
                              skipinitialspace=True,
                              index_col=0)
        units = re.findall(r'C Units:(.*?),', text)[0]

        if len(s2) > 1:
            df_cond = pd.read_csv(io.StringIO(s2[1].strip()),
                                  delim_whitespace=True,
                                  skipinitialspace=True,
                                  index_col=0)
            units_cond = re.findall(r'G Units:(.*?)\n', text)[0]
        else:
            df_cond = None

        var = re.findall(r'DesignVariation:(.*?)\n',
                         text)  # this changed circa v2020
        if len(var) < 1:  # didnt find
            var = re.findall(r'Design Variation:(.*?)\n', text)
            if len(var) < 1:  # didnt find
                # May not be present if there are no design variations to begin
                # with and no variables in the design.
                pass  #logger.error(f'Failed to parse Q3D matrix Design Variation:\nFile:{path}\nText:{text}')

                var = ['']
        design_variation = var[0]

        return df_cmat, units, design_variation, df_cond, units_cond
```

### Useful links and references

https://github.com/pyansys/pyaedt/issues/1650

Samuelopez-ansys, 2022-09-13:
@priti-ashvin-shah-ibm Could you give more info?

If I understood well you would need a method that extracts the information created by the ExportNetwork method right?

I am going to check how it is calculated and I think a new method that extracts this info would be nice right? or you just need Capacitance Matrix and Conductance Matrix?

In that case, what problem do you have with the get_solution_data method?

priti-ashvin-shah-ibm, 2022-09-14:
I created this issue since the order of data could change when you create the new method to extract the data.  In particular, I would like to have the order of table be the same as reading from the file.   That is why I shared to code which reads the file.

Samuelopez-ansys, 2022-09-14:
For this case, I think we can not change the order because get_solution_data is a method shared between ALL AEDT applications, but you could implement a ‘sort’ function between the getsolutiondata and your parser.

I think with the new pandas Pull request, you will access to the data in a more easy way.

If you don’t agree please let me know how you think it could be implemented, or if you agree please close the issue :)

Regards

closed 2022-09-14 by Samuelopez-ansys