ansys/pyaedt #1701: Bug located in PostProcessor.ofieldsreporter ClcEval call

obrienpja, 2022-09-15:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am doing a calculation using the field calculator, and I am unable to run the scripting call to evaluate the stack, ClcEval(). 

Here is the stack I am trying to evaluate (I have confirmed that I can evaluate it in the GUI.):
[image]

I am using the following PostProcessor/scripting syntax to try to evaluate the field calculator stack:
```
hfss_instance.post.ofieldsreporter.ClcEval(setup_name, ["Phase:=", "0deg"])
```

When I try running this code in my debugger, this is what it looks like:
[image]
Notice that the is_solved member variable is False.

The Ansys message panel only ever says that I can't evaluate the stack because the solution does not exist:
[image]

But when I look at the solution data for this setup, I can see that the simulation has run:
[image]

### Steps To Reproduce

1. Run an eigenmode simulation
2. Use the ofieldsreporter object on the PostProcessor of the app used to run the simulation to calculate some quantity on the stack of the field calculator (I calculated the electric energy, which was essentially the integral over some volume of the E field squared times the permittivity)
3. Call the ClcEval() method like this:
```
hfss_instance.post.ofieldsreporter.ClcEval(setup_name, ["Phase:=", "0deg"])
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

```
addict @ file:///home/conda/feedstock_root/build_artifacts/addict_1636818143388/work
aiohttp==3.8.1
aiosignal==1.2.0
alabaster==0.7.12
anyio==3.6.1
appdirs==1.4.4
argon2-cffi @ file:///home/conda/feedstock_root/build_artifacts/argon2-cffi_1640817743617/work
argon2-cffi-bindings @ file:///D:/bld/argon2-cffi-bindings_1649500527917/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1618968359944/work
async-timeout==4.0.2
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1640799537051/work
Babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1655419414885/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
backports.functools-lru-cache @ file:///home/conda/feedstock_root/build_artifacts/backports.functools_lru_cache_1618230623929/work
beautifulsoup4 @ file:///home/conda/feedstock_root/build_artifacts/beautifulsoup4_1649463573192/work
bleach @ file:///home/conda/feedstock_root/build_artifacts/bleach_1656355450470/work
branca @ file:///home/conda/feedstock_root/build_artifacts/branca_1650311966802/work
brotlipy @ file:///D:/bld/brotlipy_1648854347500/work
cached-property @ file:///home/conda/feedstock_root/build_artifacts/cached_property_1615209429212/work
certifi==2022.6.15
cffi @ file:///D:/bld/cffi_1656782940775/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1655906222726/work
click @ file:///D:/bld/click_1651215332191/work
[... 128 lines cut ...]
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1655315839047/work
statsmodels @ file:///D:/bld/statsmodels_1644535944524/work
sympy @ file:///D:/bld/sympy_1647802833311/work
terminado @ file:///D:/bld/terminado_1652790796353/work
threadpoolctl @ file:///home/conda/feedstock_root/build_artifacts/threadpoolctl_1643647933166/work
tinycss2 @ file:///home/conda/feedstock_root/build_artifacts/tinycss2_1637612658783/work
toml @ file:///home/conda/feedstock_root/build_artifacts/toml_1604308577558/work
tornado @ file:///D:/bld/tornado_1656937966227/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1649051611147/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1655411388954/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1656706066251/work
unicodedata2 @ file:///D:/bld/unicodedata2_1649112137711/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1657224465922/work
vtk==9.2.0rc1
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1600965781394/work
webencodings==0.5.1
websocket-client==1.3.3
widgetsnbextension @ file:///home/conda/feedstock_root/build_artifacts/widgetsnbextension_1655939017940/work
win-inet-pton @ file:///D:/bld/win_inet_pton_1648771897787/work
wslink==1.6.6
xarray @ file:///home/conda/feedstock_root/build_artifacts/xarray_1646254587173/work
xyzservices @ file:///home/conda/feedstock_root/build_artifacts/xyzservices_1655861233352/work
yarl==1.7.2
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1649012893348/work
```

obrienpja, 2022-09-15:
Hi @Samuelopez-ansys, can you have a look at this, please?

Samuelopez-ansys, 2022-09-15:
Hi @obrienpja ,

Please could check if your variable setup_name has the 'LastAdaptive' part?

In your case, it should contain: "QHFSSEigenmode_Pyaedt_setup : LastAdaptive"

And the command also needs a 3rd parameter "Fields". So the command should be:

ClcEval("QHFSSEigenmode_Pyaedt_setup : LastAdaptive", ["Phase:=", "0deg"], "Fields")

It is not a pyaedt issue, it is AEDT API :(

Let me know if it works.

obrienpja, 2022-09-15:
Hi @Samuelopez-ansys, thank you for taking a look at this! I tried the syntax you posted here, and I am getting the same error:
```
self.current_app.post.ofieldsreporter.ClcEval("QHFSSEigenmode_Pyaedt_setup : LastAdaptive", ["Phase:=", "0deg"], "Fields")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<COMObject GetModule>", line 2, in ClcEval
  File "C:\Anaconda3\envs\pob_qiskit_metal_pyaedt2\lib\site-packages\win32com\client\dynamic.py", line 368, in _ApplyTypes_
    result = self._oleobj_.InvokeTypes(
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024365), None)
```

Ansys also still says the solution does not exist. I tried a couple other variants of the syntax given your suggestion, and they all give the same error.

Samuelopez-ansys, 2022-09-16:
Hi @obrienpja ,

Really weird, because it is not a pyaedt command here you are using directly an AEDT API command ("ClcEval").

Please try to Record the script in AEDT, go to Fields Calculator and Push Eval. If it works, please check the recorded script, the parameters of the command should be the same. If does not work, it means the design is not solved.

obrienpja, 2022-09-25:
Hi @Samuelopez-ansys,
I checked again, and the syntax you gave is correct. I had an extra underscore in my setup name. Thank you very much for helping with this!

closed 2022-09-25 by obrienpja