ansys/pyaedt #1704: Remove dependencies on libraries in $ANSYSEM_ROOT_PATH/Delcross or remove warning about this folder

isaacansys, 2022-09-15:
### Description of the feature

If $ANSYSEM_ROOT_PATH/Delcross is not in $LD_LIBRARY_PATH on Linux, PyAEDT will issue a warning.  This folder contains, among other libraries, Qt 5 libraries that may prevent the use of later Qt 5 versions by downstream apps that depend on PyAEDT.  In testing with such an app I found that at least EDB-only usage of PyAEDT seems to work without the Delcross folder in $LD_LIBRARY_PATH.

Samuelopez-ansys, 2022-11-11:
@MaxJPRey What do you think about this issue?

maxcapodi78, 2023-04-17:
this may be probably removed in 23R2. I need to run some checks in linux

PR #3634 (2023-09-26): Remove Delcross from reference
Fix #1704
closed 2023-09-26 by maxcapodi78