ansys/pyaedt #1708: Problem of clr_loader upgrade

OpenActuator, 2022-09-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

PyAEDT does not work because of clr_loader upgrade in pythonnet package.

The error message is "AttributeError: module 'clr_loader' has no attribute 'wrappers'" as shown below.

[image]

I found the reason of above problem.
The reason was the clr_loader upgrade on Sep 17, 2022. ( https://pypi.org/project/clr-loader/#history )

### Steps To Reproduce

I temporarily solved the issue by downgrading clr_loader to 0.1.7.

- pip uninstall clr_loader
- pip install clr_loader==0.1.7

Please solve PyAEDT for the upgrade of clr_loader.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.3
contourpy==1.0.5
cycler==0.11.0
fonttools==4.37.2
frozenlist==1.3.1
idna==3.4
imageio==2.21.3
kiwisolver==1.4.4
matplotlib==3.6.0
multidict==6.0.2
numpy==1.23.3
packaging==21.3
pandas==1.4.4
Pillow==9.2.0
plumbum==1.7.2
psutil==5.9.2
pyaedt==0.6.2
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.0rc4
pytz==2022.2.1
pyvista==0.36.1
pywin32==304
rpyc==5.0.1
scooby==0.6.0
six==1.16.0
vtk==9.2.0rc2
wslink==1.8.2
yarl==1.8.1

closed 2022-09-19 by Samuelopez-ansys

MaxJPRey, 2022-09-20:
After further testing, I was also able to increase the version for Pythonnet.

**Note:** it is still an 'rc' version

OpenActuator, 2022-09-21:
I checked it.
I made a mistake.
It has already been applied, but I didn't know it had been released yet.

I believe pyaedt will help us a lot.
Thank you.

MaxJPRey, 2022-09-21:
You are welcome.
Thanks for your feedback. It is much appreciated and valuable for us.

Glad to hear that you don't face this issue anymore.

Best Regards.