ansys/pyaedt #1716: Lambda refinement ignored in SIwave Advanced solver settings

mahesh-dhanekula, 2022-09-20:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

DoLambdaRefinement feature of configuration file needs to be considered for letting user to control the meshing SIwaveSYZ solver.

### Steps To Reproduce

Run the build_simulation_project() using sample configuration file. Observe the Meshing property in SIwaveSYZ solver advanced settings through UI.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

Tested with virtual environment created using python 3.9 where pyaedt and its dependencies are the only installed packages

maxcapodi78, 2023-04-17:
@svandenb-dev please check if this is still a bug

svandenb-dev, 2023-04-17:
this was not a bug and has been clarified with customer. closing this issue