ansys/pyaedt #1737: Create circuit port or source from pin group

pyAnsysUser, 2022-09-26:
### Description of the feature

The documentation outlines how to create pin groups as well as voltage/current sources and circuit ports, but there does not appear to be a way to use the created pin group in the creation of the source or circuit port.

ring630, 2022-10-31:
@pyAnsysUser new features are implemented. Please let me know if this is what you expected.
        
            pg_name, _ = edbapp.core_siwave.create_pin_group("U3A1", 2)
            edbapp.core_siwave.create_pin_group("U3A1", [5, 34, 35], "pos")
            edbapp.core_siwave.create_pin_group_on_net("U3A1", "GND", "gnd")
            edbapp.core_siwave.create_current_source_on_pin_group("pos", "gnd")
            edbapp.core_siwave.create_voltage_source_on_pin_group(pg_name, "gnd")

ring630, 2022-10-31:
@pyAnsysUser , I am closing this issue. Please feel free to submit a new issue if you wants new methods.

closed 2022-10-31 by ring630