ansys/pyaedt #1750: Bug located in ...Advanced FAr Field postprocessing example

arobinson905, 2022-09-28:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The example shown on the pyansys website 
(url here: https://aedt.docs.pyansys.com/examples/02-HFSS/Advanced_Far_Field.html#sphx-glr-examples-02-hfss-advanced-far-field-py) is broken. 

The error that the plotting function throws is published on the PyAnsys webpage as a "result".
[image]

One possible fix to this plotting function is to explicitly call the last adaptive pass rather than use the nominal function call 
vals = hfss.post.get_far_field_data(
    setup_sweep_name="Setup1 : LastAdaptive", expression="RealizedGainTotal", domain="Elevation"
)
instead of:    #setup_sweep_name=hfss.nominal_sweep, expression="RealizedGainTotal", domain="Elevation"

however, if you are not using the Spyder compiler, this function automatically closes all plots it creates, so it is useless to the user anyway.

The function setup_sweep_name=hfss.nominal_sweep is broken because this example of a finite array does not create a solved variation to apply to be called "nominal". I am unsure how this script broke HFSS. If you navigate in HFSS to HFSS>>Results>>Apply Solved Variation, HFSS tells you that no solved variations exist. I do not know how to fix this. 

Python 3.10 does not support this script, you have to use Python 3.9

### Steps To Reproduce

1) Run the example script "Advanced Far Field Postprocessing"  from the website (https://aedt.docs.pyansys.com/examples/02-HFSS/Advanced_Far_Field.html#sphx-glr-examples-02-hfss-advanced-far-field-py)

2) Observe the error that is published on the website as a "result"

3) replace all function calls of "setup_sweep_name=hfss.nominal_sweep" with "setup_sweep_name="Setup1 : LastAdaptive"

4) errors are removed, but no plots are created unless you use Spyder complier

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
alabaster==0.7.12
appdirs==1.4.4
arrow==1.2.3
astroid==2.12.10
async-timeout==4.0.2
atomicwrites==1.4.1
attrs==22.1.0
autopep8==1.6.0
Babel==2.10.3
backcall==0.2.0
bcrypt==4.0.0
beautifulsoup4==4.11.1
binaryornot==0.4.4
black==22.8.0
bleach==5.0.1
CacheControl==0.12.11
certifi==2022.9.24
cffi==1.15.1
chardet==5.0.0
charset-normalizer==2.1.1
click==8.1.3
cloudpickle==2.2.0
clr-loader==0.2.3
colorama==0.4.5
contourpy==1.0.5
cookiecutter==2.1.1
cryptography==38.0.1
cycler==0.11.0
debugpy==1.6.3
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20200713
dill==0.3.5.1
docutils==0.19
entrypoints==0.4
fastjsonschema==2.16.2
flake8==4.0.1
fonttools==4.37.3
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
imagesize==1.4.1
importlib-metadata==4.12.0
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.16.0
ipython==7.34.0
ipython-genutils==0.2.0
isort==5.10.1
jaraco.classes==3.2.2
jedi==0.18.1
jellyfish==0.9.0
Jinja2==3.1.2
jinja2-time==0.2.0
jsonschema==4.16.0
jupyter-core==4.11.1
jupyter_client==7.3.5
jupyterlab-pygments==0.2.2
keyring==23.9.3
kiwisolver==1.4.4
lazy-object-proxy==1.7.1
lxml==4.9.1
MarkupSafe==2.1.1
matplotlib==3.6.0
matplotlib-inline==0.1.6
mccabe==0.6.1
mistune==2.0.4
more-itertools==8.14.0
msgpack==1.0.4
multidict==6.0.2
mypy-extensions==0.4.3
nbclient==0.6.8
nbconvert==7.0.0
nbformat==5.6.1
nest-asyncio==1.5.5
numpy==1.23.3
numpydoc==1.4.0
packaging==21.3
pandas==1.5.0
pandocfilters==1.5.0
paramiko==2.11.0
parso==0.8.3
pathspec==0.10.1
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.2.0
pip-api==0.0.30
pip-download==0.4.2
platformdirs==2.5.2
pluggy==1.0.0
plumbum==1.7.2
prompt-toolkit==3.0.31
psutil==5.9.2
ptyprocess==0.7.0
pyaedt==0.6.4
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.13.0
pylint==2.15.3
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.7
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.0
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
pyrsistent==0.18.1
python-dateutil==2.8.2
python-lsp-black==1.2.1
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.5.0
python-slugify==6.1.2
pythonnet==3.0.0rc6
pytoolconfig==1.2.2
pytz==2022.2.1
pytz-deprecation-shim==0.1.0.post0
pyvista==0.36.1
pywin32==304
pywin32-ctypes==0.2.0
PyYAML==6.0
pyzmq==24.0.1
QDarkStyle==3.0.3
qstylizer==0.2.2
QtAwesome==1.1.1
qtconsole==5.3.2
QtPy==2.2.0
requests==2.28.1
retrying==1.3.3
rope==1.3.0
rpyc==5.0.1
Rtree==1.0.0
scooby==0.6.0
six==1.16.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.3.2.post1
Sphinx==5.2.1
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder==5.3.3
spyder-kernels==2.3.3
text-unidecode==1.3
textdistance==4.5.0
three-merge==0.1.1
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.4
tornado==6.2
traitlets==5.4.0
typing_extensions==4.3.0
tzdata==2022.4
tzlocal==4.2
ujson==5.5.0
urllib3==1.26.12
vtk==9.1.0
watchdog==2.1.9
wcwidth==0.2.5
webencodings==0.5.1
whatthepatch==1.0.2
wrapt==1.14.1
wslink==1.8.2
yapf==0.32.0
yarl==1.8.1
zipp==3.8.1

arobinson905, 2022-09-28:
[image]
[image]

Samuelopez-ansys, 2022-09-30:
Hi @arobinson905 ,

I found the bug. The issue happens when you define a Discrete sweep with more than 1 frequency. Then in vals3d is saved all the data (76GHz and 77 GHz) in this case, and the method plot_3d() was not ready for that.

I have fixed it and now if you call this method , it will plot the active frequency. FYI, when you load any data with pyaedt like:

_vals3d1 = hfss.post.get_far_field_data(
    setup_sweep_name=hfss.nominal_sweep, expression="RealizedGainTotal", domain="Infinite Sphere1"
)_

There is an active variation:

_vals3d1.active_intrinsic_

If you want to change it you just need:
_vals3d1.active_intrinsic["Freq"] = 77.0_

And then you could plot the data at 77GHz:
_vals3d1.plot_3d()_

closed 2022-09-30 by maxcapodi78

arobinson905, 2022-09-30:
Fantastic! Thank you so much!

[External Sender]

PR #1768 (2022-09-30): Farfield 3D plot with more than 1 frequency sweep
Fix bug when frequency sweep has more than 1 frequency and the user wants to plot the 3D FF.

Close #1750