ansys/pyaedt #1838: Q2D sweep props['SaveRadFields']= True does't check the box for a Discrete Sweep

cmesibov, 2022-10-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

This code I expected to set the analysis setup and sweep.  Saving fields only applies to discrete sweeps.  The code below attempts to do that and select the Save Radiated Fields check box.  But the checkbox is not being selected.

from pyaedt import Q2d

q = Q2d(designname="Cable_model")
qSetup = q.create_setup("Setup1")
qSetup.props["AdaptiveFreq"] = "1GHz"
qSetup.update()
qSweep = qSetup.add_sweep("sweep1")
qSweep.props["Type"] = "Discrete"
qSweep.props["RangeType"] = "LinearCount"
qSweep.props["RangeStart"] = 0
qSweep.props["RangeEnd"] = 0.1
qSweep.props["RangeStep"] = 10
qSweep.props['SaveRadFields'] = True
qSweep.add_subrange("LinearStep", 0.1, 1, 0.050)
qSweep.update()
qSetup.update()

### Steps To Reproduce

1) Use the above code and examine the resultant sweep settings verify the Save Radiated Fields not checked 

### Which Operating System are you using?

Windows

### Which Python version are you using?

IronPython

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.11.2
async-timeout==4.0.2
asynctest==0.13.0
atomicwrites==1.4.0
attrs==21.4.0
autopep8==1.6.0
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
cffi==1.15.1
cfgv==3.3.1
charset-normalizer==2.0.12
colorama==0.4.4
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.4
distlib==0.3.4
entrypoints==0.4
fastjsonschema==2.15.3
filelock==3.6.0
flake8==4.0.1
fonttools==4.29.1
frozenlist==1.3.0
identify==2.5.0
idna==3.3
imageio==2.16.0
importlib-metadata==4.11.3
importlib-resources==5.8.0
iniconfig==1.1.1
ipykernel==6.15.1
ipython==7.31.1
ipython-genutils==0.2.0
ipywidgets==7.7.1
isort==5.10.1
jedi==0.18.1
Jinja2==3.1.2
jsonschema==4.7.2
jupyter==1.0.0
jupyter-client==7.3.4
jupyter-console==6.4.4
jupyter-core==4.11.1
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.1
kiwisolver==1.3.2
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.5.1
matplotlib-inline==0.1.3
mccabe==0.6.1
mistune==0.8.4
multidict==6.0.2
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
nodeenv==1.6.0
notebook==6.4.12
numpy==1.21.5
packaging==21.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.0.1
platformdirs==2.5.1
pluggy==1.0.0
pre-commit==2.18.1
prometheus-client==0.14.1
prompt-toolkit==3.0.28
psutil==5.9.1
py==1.11.0
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.11.2
pylint==2.13.5
pyparsing==3.0.7
pyrsistent==0.18.1
PySide6==6.2.4
pytest==7.1.2
python-dateutil==2.8.2
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.4.1
pyvista==0.33.2
pywin32==304
pywinpty==2.0.6
PyYAML==6.0
pyzmq==23.2.0
qtconsole==5.3.1
QtPy==2.1.0
rope==1.0.0
scooby==0.5.11
Send2Trash==1.8.0
shiboken6==6.2.4
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.3.2.post1
terminado==0.15.0
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tornado==6.2
traitlets==5.3.0
typed-ast==1.5.2
typing_extensions==4.1.1
ujson==5.2.0
virtualenv==20.14.1
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
widgetsnbextension==3.6.1
wrapt==1.14.0
wslink==1.4.1
yapf==0.32.0
yarl==1.7.2
zipp==3.8.0

cmesibov, 2022-10-28:
Found that the property is 'SaveFields' not 'SaveRadFields'.  Works with 'SaveFields'

closed 2022-10-28 by cmesibov

Samuelopez-ansys, 2023-12-11:
@cmesibov Nevermind, I have just assigned to you because you closed it. But it is solved.