ansys/pyaedt #1848: Circuit  assign_voltage_frequency_dependent_excitation_to_ports, Only allows for one frequency dependant source

cmesibov, 2022-10-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

oDesign.UpdateSources in the "NAME:Data" list allows for multiple voltage frequency dependent sources.  The pyaedt implementation only allows for one voltage source in the data list.  When iterating over model pins and assigning unique spectral data for that pin, this causes any previous assigned FDS data to be overwritten by the new source.
Although the function allows for one fds files to be assigned to multiple pins, it is most likely that there would be multiple fds/pin assignments.

It seems that the function needs to read the sources prior to populating the list.  Or a separate script that reads the source as "arg1" data and have it as a optional command argument. 

### Steps To Reproduce

1) Examine circuit.py assign_voltage_frequency_dependent_excitation_to_ports starting at line 1437.
2) Starting at line 1472 the data list is fixed

### Which Operating System are you using?

Windows

### Which Python version are you using?

IronPython

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.11.2
async-timeout==4.0.2
asynctest==0.13.0
atomicwrites==1.4.0
attrs==21.4.0
autopep8==1.6.0
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
cffi==1.15.1
cfgv==3.3.1
charset-normalizer==2.0.12
colorama==0.4.4
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.4
distlib==0.3.4
entrypoints==0.4
fastjsonschema==2.15.3
filelock==3.6.0
flake8==4.0.1
fonttools==4.29.1
frozenlist==1.3.0
identify==2.5.0
idna==3.3
imageio==2.16.0
importlib-metadata==4.11.3
importlib-resources==5.8.0
iniconfig==1.1.1
ipykernel==6.15.1
ipython==7.31.1
ipython-genutils==0.2.0
ipywidgets==7.7.1
isort==5.10.1
jedi==0.18.1
Jinja2==3.1.2
jsonschema==4.7.2
jupyter==1.0.0
jupyter-client==7.3.4
jupyter-console==6.4.4
jupyter-core==4.11.1
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.1
kiwisolver==1.3.2
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.5.1
matplotlib-inline==0.1.3
mccabe==0.6.1
mistune==0.8.4
multidict==6.0.2
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
nodeenv==1.6.0
notebook==6.4.12
numpy==1.21.5
packaging==21.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.0.1
platformdirs==2.5.1
pluggy==1.0.0
pre-commit==2.18.1
prometheus-client==0.14.1
prompt-toolkit==3.0.28
psutil==5.9.1
py==1.11.0
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.11.2
pylint==2.13.5
pyparsing==3.0.7
pyrsistent==0.18.1
PySide6==6.2.4
pytest==7.1.2
python-dateutil==2.8.2
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.4.1
pyvista==0.33.2
pywin32==304
pywinpty==2.0.6
PyYAML==6.0
pyzmq==23.2.0
qtconsole==5.3.1
QtPy==2.1.0
rope==1.0.0
scooby==0.5.11
Send2Trash==1.8.0
shiboken6==6.2.4
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.3.2.post1
terminado==0.15.0
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tornado==6.2
traitlets==5.3.0
typed-ast==1.5.2
typing_extensions==4.1.1
ujson==5.2.0
virtualenv==20.14.1
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
widgetsnbextension==3.6.1
wrapt==1.14.0
wslink==1.4.1
yapf==0.32.0
yarl==1.7.2
zipp==3.8.0

Samuelopez-ansys, 2022-10-28:
Hi @cmesibov ,

This is a current limitation in PyAEDT. I submitted an issue months ago (#1537 ) But the solution is tricky. For your specific project, do you need only Frequenecy dependent excitations (with different .fds files)? Or any other type of excitation?

cmesibov, 2022-10-28:
Hi Sam,
I have coded my own version of this function that gets the job done.  I am happy to share it with you.  I just don't have the time now to add it to pyaedt.
It uses 3 data structures as inputs and updates all the ports accordingly.
Chris

[External Sender]

PR #2157 (2022-12-20): New Circuit Source and Excitation class
New classes to allow full control of Excitations and Sources in Circuit. 

Close #1848 #1537